MiCode/Xiaomi_Kernel_OpenSource #1202: umi-q-oss source does not build: make[3]: *** No rule to make target 'arch/arm64/boot/dts/qcom/apq8016-sbc.dtb', needed by '__build'. Stop.

mrslezak, 2020-02-27:
Source for umi-q-oss does not build.  make[3]: *** No rule to make target 'arch/arm64/boot/dts/qcom/apq8016-sbc.dtb', needed by '__build'. Stop.

The compilation can continue only if all entries for: CONFIG_ARCH_QCOM in arch/arm64/boot/dts/qcom/Makefile are commented out.  This happens with the stock repo, with GCC10, with Clang, no matter the compiler.  

After merging all portions of the Qualcomm source in properly (techpack audio, data_kernel, qcacld wlan, device_tree), the build error remains.

Merged code is here: https://github.com/mrslezak/umi-q-oss, A debug build as well (kernel.log) and a build script "Build GCC10 Umi.sh." I have consulted with several professional developers and they cannot resolve the issue.  Whatever source code changes Xiaomi made internally to compile were not pushed to the repo, or the repo version was not tested before posting.

Please review and remedy with commits. Otherwise, this source is not useable for anyone.  I have spent many development hours trying to get a Mi10 / Mi10 Pro kernel to work, and there are countless issues that just cannot be solved by one person.

Aquarius223, 2020-03-01:
You can try to delete the following strings in "arch / arm64 / boot / dts / qcom / Makefile"
"dtb-$ (CONFIG_ARCH_QCOM) + = apq8016-sbc.dtb" It is the device tree of Snapdragon 410, our Snapdragon 865 does not need it

LenHackintosh, 2020-03-01:
> You can try to delete the following strings in "arch / arm64 / boot / dts / qcom / Makefile"
> "dtb-$ (CONFIG_ARCH_QCOM) + = apq8016-sbc.dtb" It is the device tree of Snapdragon 410, our Snapdragon 865 does not need it

it won't work..just a issue with custom dtc.

mrslezak, 2020-03-01:
Dev buddy of mine said it build find with Clang11 on his Mint Linux box.  So I think it's my Ubuntu 18.04 VM.  Another guy had DTB issues on Ubuntu and switched to Arch Linux and no problem.  Just waiting for confirmation.  What it is on Ubuntu causing 4.19 source to fail on DTB compiling, I have no idea.  I built with Clang11 and it failed same error on Ubuntu.

mrslezak, 2020-03-02:
Issue is resolved.  Don't use Ubuntu or apply the fix to Makefile.lib in my repo.

closed 2020-03-02 by mrslezak

TheGlitchh, 2020-05-15:
> > You can try to delete the following strings in "arch / arm64 / boot / dts / qcom / Makefile"
> > "dtb-$ (CONFIG_ARCH_QCOM) + = apq8016-sbc.dtb" It is the device tree of Snapdragon 410, our Snapdragon 865 does not need it
> 
> it won't work..just a issue with custom dtc.

okay so how to fix it ?

a007XDA, 2022-07-22:
"sudo apt-get install device-tree-compiler" should do it.

Henri-J-Norden, 2022-10-22:
> Issue is resolved. Don't use Ubuntu or Debian or apply the fix to Makefile.lib in my repo. Thanks goes to LenHackintosh

Would have been nice link the patch or at least the repo... 
[I found it here](https://github.com/mrslezak/venus-r-oss/commit/bdfe55128ff56d338daa18793aeb384e6f14326c#diff-1d425d4979d94f4c641e85ac1408dd134a0ce69a1db593f2ea9aa4015fe0c0d3): [git patch](https://github.com/mrslezak/venus-r-oss/commit/bdfe55128ff56d338daa18793aeb384e6f14326c.patch)

```diff
diff --git a/scripts/Makefile.lib b/scripts/Makefile.lib
index e62bb0513297..3a402daddcab 100644
--- a/scripts/Makefile.lib
+++ b/scripts/Makefile.lib
@@ -290,7 +290,7 @@ cmd_dtc = mkdir -p $(dir ${dtc-tmp}) ; \
 		-d $(depfile).dtc.tmp $(dtc-tmp) ; \
 	cat $(depfile).pre.tmp $(depfile).dtc.tmp > $(depfile)
 
-$(obj)/%.dtb: $(src)/%.dts $(DTC) FORCE
+$(obj)/%.dtb: $(src)/%.dts FORCE
 	$(call if_changed_dep,dtc,dtb,-@)
 
 DT_CHECKER ?= dt-validate
```

I had the same issue compiling `cas-q-oss` on Ubuntu 22.04 (WSL2), although I eventually realized that it is also missing all of the kona/SM8250 device tree files for cas and I found [this maintained repo by UtsavBalar1231 for SM8250 devices](https://github.com/UtsavBalar1231/kernel_xiaomi_sm8250)

Uzver123, 2025-02-10:
> "`sudo apt-get install device-tree-compiler`" should do it.

This error only shows **external device-tree-compiler** module is used. 

Error show only when specified argument `DTC_EXT=dtc` or `DTC=dct` (which simply points to **dtc** executable e.g. `/usr/bin/dtc`).
If not specified it will use kernel built-in DTC compiler, which probably is more correct version. 

p.s. For old Kernel v4.19 external DTC compiler version should be <= v1.6

Uzver123, 2025-02-10:
> "`sudo apt-get install device-tree-compiler`" should do it.

Error only shows when **external device-tree-compiler** executable is used for compiling `.dts` to `.dtb`.

Precisely error shows when argument `DTC_EXT=dtc` or `DTC=dct` (which simply points to **dtc** executable`) is specified and is not pointing correctly to executable. So simple solution if you want to use external tool then set `DTC_EXT=/usr/bin/dtc`

When argument `DTC_EXT=dtc` or `DTC=dct` not specified it will use kernel built-in DTC compiler, which probably is more correct version. 

p.s. For kernel v4.19 external DTC compiler version should be <= v1.6

Uzver123, 2025-02-10:
> "`sudo apt-get install device-tree-compiler`" should do it.

Error only shows when **external device-tree-compiler** executable is used for compiling `.dts` to `.dtb`.

Precisely error shows when argument `DTC_EXT=dtc` or `DTC=dct` (which simply points to **dtc** executable) is specified and is not pointing correctly to executable. Simple solution if you want to use external tool then set path properly e.g. `DTC_EXT=/usr/bin/dtc`

When argument `DTC_EXT=dtc` or `DTC=dct` not specified it will use kernel built-in DTC compiler, which probably is more correct version. 

P.S. For kernel v4.19 external DTC compiler version should be <= v1.6

Uzver123, 2025-02-10:
> "`sudo apt-get install device-tree-compiler`" should do it.

Error only shows when **external device-tree-compiler** executable is used for compiling `.dts` to `.dtb`.

Precisely error shows when argument `DTC_EXT=dtc` (which simply points to **dtc** executable) is specified and is not pointing correctly to executable. Simple solution if you want to use external tool then set path properly e.g. `DTC_EXT=/usr/bin/dtc`

When argument `DTC_EXT=dtc` not specified it will use kernel built-in DTC compiler, which probably is more correct version. 

P.S. For kernel v4.19 external DTC compiler version should be <= v1.6

Uzver123, 2025-02-10:
> "`sudo apt-get install device-tree-compiler`" should do it.

Error only shows when **external device-tree-compiler** executable is used for compiling `.dts` to `.dtb`.

Precisely error shows when argument `DTC_EXT=dtc` (which simply points to **dtc** executable) is specified and is not pointing correctly to executable. Simple solution if you want to use external tool then set path properly e.g. `DTC_EXT=/usr/bin/dtc`

When argument `DTC_EXT` not specified it will use kernel built-in DTC compiler, which probably is more correct version. 

P.S. For kernel v4.19 external DTC compiler version should be <= v1.6

Uzver123, 2025-02-11:
> "`sudo apt-get install device-tree-compiler`" should do it.

Error only shows when **external device-tree-compiler** executable is used for compiling `.dts` to `.dtb`.

Precisely error shows when argument `DTC_EXT=dtc` (which simply points to **dtc** executable) is specified and is not pointing correctly to executable. Simple solution if you want to use external tool then set path properly e.g. `DTC_EXT=/usr/bin/dtc`

When argument `DTC_EXT` not specified it will use kernel built-in DTC compiler, which probably is more correct version. 

P.S. For kernel v4.19 you should be using external [DTC](https://snapshot.debian.org/package/device-tree-compiler/1.4.4-1/) version 1.4.4-1 (2017) as newer versions give warnings for cas-q-oss

Uzver123, 2025-02-11:
> "`sudo apt-get install device-tree-compiler`" should do it.

Surprised how people are in hurry to apply questionable fixes when problem is very simple.

`DTC_EXT=dtc` is option that points to executable of external **device-tree-compiler** (it parses `.dts` files to `.dtb`) 
Error only shows if you used option `DTC_EXT` and did not provide correct path to executable which is `/usr/bin/dtc`. 
To avoid error simply use `DTC_EXT=/usr/bin/dtc` or specify option DTC_EXT at all then it will automatically be found by Kernel.

When argument `DTC_EXT` not specified Kernel will built-in it from its sources (DTC v1.4.7-x).

If you want to use external DTC for kernel v4.19 you should be using external [DTC](https://snapshot.debian.org/package/device-tree-compiler/) version v1.4.6-1 or newer:

- DTC v1.4.6-1 (lowest one that support this syntax)
- DTC v1.4.7-x (same as built into the kernel)
- DTC v1.5.1-1 (highest that installs on Ubuntu 20) 

DTC throw lots of warnings, but compilations works, both this versions v1.4.6-1 and DTC v1.5.1-1 produce same `.dts`, however  v1.4.7-x output `.dts` is different, if one does not work you can try another version.

Uzver123, 2025-02-11:
> "`sudo apt-get install device-tree-compiler`" should do it.

Surprised how people are in hurry to apply questionable fixes when problem is very simple.

`DTC_EXT=dtc` is option that points to executable of external **device-tree-compiler** (it parses `.dts` files to `.dtb`) 
Error only shows if used option `DTC_EXT` and did not provide correct path to executable which is `/usr/bin/dtc`. 
To avoid error simply use `DTC_EXT=/usr/bin/dtc` or dont specify option DTC_EXT at all then it will automatically be found by Kernel.

When option `DTC_EXT` not specified Kernel will built DTC from included sources (DTC v1.4.7-x).

If you want to use external DTC for kernel v4.19 you should be using external [DTC](https://snapshot.debian.org/package/device-tree-compiler/) version v1.4.6-1 or newer:

- DTC v1.4.6-1 (lowest one that support this syntax)
- DTC v1.4.7-x (same as built into the kernel)
- DTC v1.5.1-1 (highest that installs on Ubuntu 20) 

DTC throw lots of warnings, but compilations works, both this versions v1.4.6-1 and DTC v1.5.1-1 produce same `.dts`, however  v1.4.7-x output `.dts` is different, if one does not work you can try another version.

Uzver123, 2025-02-11:
> "`sudo apt-get install device-tree-compiler`" should do it.

Surprised how people are in hurry to apply questionable fixes when problem is very simple.

`DTC_EXT=dtc` is option that points to executable of external **device-tree-compiler** (it parses `.dts` files to `.dtb`) 
Error only shows if used option `DTC_EXT` and did not provide correct path to executable which is `/usr/bin/dtc`. 
To avoid error simply use `DTC_EXT=/usr/bin/dtc` or dont specify option DTC_EXT at all then it will automatically be found by Kernel.

When option `DTC_EXT` not specified Kernel will built DTC from included sources (DTC v1.4.7-x).

For kernel v4.19 should be using external [DTC](https://snapshot.debian.org/package/device-tree-compiler/) v1.4.6-1 or newer:

- DTC v1.4.6-1 (lowest one that support this syntax)
- DTC v1.4.7-x (same as built into the kernel)
- DTC v1.5.1-1 (highest that installs on Ubuntu 20) 

DTC throw lots of warnings, but compilations works, both this versions v1.4.6-1 and DTC v1.5.1-1 produce same `.dts`, however  v1.4.7-x output `.dts` is different, if one does not work you can try another version.

Uzver123, 2025-02-11:
> "`sudo apt-get install device-tree-compiler`" should do it.

Surprised how people are in hurry to apply questionable fixes when problem is very simple.

`DTC_EXT=dtc` is option that points to executable of external **device-tree-compiler** (it parses `.dts` files to `.dtb`) 
Error only shows if used option `DTC_EXT` and did not provide correct path to executable which is `/usr/bin/dtc`. 
To avoid error simply use `DTC_EXT=/usr/bin/dtc` or dont specify option DTC_EXT at all then it will automatically be found by Kernel.

When option `DTC_EXT` not specified Kernel will built DTC from included sources (DTC v1.4.7-x).

For kernel v4.19 should be using external [DTC](https://snapshot.debian.org/package/device-tree-compiler/) v1.4.6-1 or newer:

- DTC v1.4.6-1 (lowest one that does not throw syntax error after importing Xiaomi device tree from neighboring repository)
- DTC v1.4.7-x (same as built into the kernel)
- DTC v1.5.1-1 (highest that installs on Ubuntu 20) 

DTC throw lots of warnings, but compilations works, both versions v1.4.6-1 and DTC v1.5.1-1 produce same `.dts`, however  v1.4.7-x output `.dts` is different. If one does not work can try another version.