MiCode/Xiaomi_Kernel_OpenSource #1204: 编译k20pro的内核时出错

MCredbear, 2020-03-06:
CHK     include/config/kernel.release
  GEN     ./Makefile
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  Using .. as source for kernel
  CHK     scripts/mod/devicetable-offsets.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    ../scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  AR      sound/soundcore.o
  CC      kernel/time/posix-cpu-timers.o
  CC      drivers/bus/mhi/controllers/mhi_arch_qcom.o
  CC      kernel/time/posix-clock.o
  CC      fs/f2fs/inode.o
  AR      sound/usb/snd-usb-audio.o
  AR      sound/usb/snd-usbmidi-lib.o
  CC      sound/usb/usb_audio_qmi_svc.o
  AR      sound/x86/built-in.o
  CC      sound/last.o
../scripts/Makefile.build:361: recipe for target 'kernel/time/posix-cpu-timers.o' failed
  CC      drivers/bus/mhi/core/mhi_init.o
../scripts/Makefile.build:676: recipe for target 'kernel/time' failed
/home/redbear/Xiaomi_Kernel_OpenSource-cepheus-q-oss/Makefile:1124: recipe for target 'kernel' failed
  AR      drivers/cdrom/built-in.o
  CC      fs/f2fs/namei.o
../scripts/Makefile.build:361: recipe for target 'drivers/bus/mhi/controllers/mhi_arch_qcom.o' failed
../scripts/Makefile.build:676: recipe for target 'drivers/bus/mhi/controllers' failed
  CC      drivers/bus/mhi/core/mhi_main.o
  CC      fs/fat/cache.o
  CC      drivers/char/mem.o
../scripts/Makefile.build:361: recipe for target 'fs/f2fs/inode.o' failed
  CC      fs/fat/dir.o
  CC      drivers/char/random.o
../scripts/Makefile.build:361: recipe for target 'sound/usb/usb_audio_qmi_svc.o' failed
../scripts/Makefile.build:676: recipe for target 'sound/usb' failed
/home/redbear/Xiaomi_Kernel_OpenSource-cepheus-q-oss/Makefile:1124: recipe for target 'sound' failed
  CC      drivers/char/misc.o
  AR      drivers/char/agp/built-in.o
  CC      fs/fat/fatent.o
  CC      drivers/char/diag/diagchar_core.o
../scripts/Makefile.build:676: recipe for target 'fs/f2fs' failed
  CC      drivers/char/diag/diagchar_hdlc.o
  AR      drivers/char/ipmi/built-in.o
  CC      drivers/clk/clk-devres.o
  CC      drivers/bus/mhi/core/mhi_pm.o
  CC      fs/fat/file.o
  CC      fs/fat/inode.o
  CC      drivers/char/diag/diagfwd.o
  CC      drivers/clk/clk-bulk.o
  CC      fs/fat/misc.o
  CC      fs/fat/nfs.o
  CC      drivers/clk/clkdev.o
  CC      drivers/char/hw_random/core.o
  CC      fs/fat/namei_vfat.o
  CC      fs/fat/namei_msdos.o
  CC      drivers/bus/mhi/core/mhi_boot.o
  CC      drivers/char/diag/diagfwd_peripheral.o
  CC      drivers/char/diag/diagfwd_socket.o
  CC      drivers/clk/clk.o
  CC      drivers/char/hw_random/msm_rng.o
  AR      fs/fat/fat.o
  AR      fs/fat/msdos.o
  CC      drivers/char/adsprpc.o
  AR      fs/fat/vfat.o
  AR      fs/fat/built-in.o
/home/redbear/Xiaomi_Kernel_OpenSource-cepheus-q-oss/Makefile:1124: recipe for target 'fs' failed
  CC      drivers/char/diag/diagfwd_rpmsg.o
  CC      drivers/char/hw_random/cavium-rng.o
  CC      drivers/char/diag/diag_mux.o
  CC      drivers/bus/mhi/core/mhi_dtr.o
  CC      drivers/clocksource/timer-of.o
  CC      drivers/char/diag/diag_memorydevice.o
  CC      drivers/char/hw_random/cavium-rng-vf.o
  CC      drivers/char/diag/diag_usb.o
  CC      drivers/clocksource/timer-probe.o
  CC      drivers/char/diag/diagmem.o
  AR      drivers/bus/mhi/core/built-in.o
../scripts/Makefile.build:676: recipe for target 'drivers/bus/mhi' failed
../scripts/Makefile.build:676: recipe for target 'drivers/bus' failed
  CC      drivers/char/adsprpc_compat.o
  AR      drivers/char/hw_random/rng-core.o
  AR      drivers/char/hw_random/built-in.o
  CC      drivers/char/fastcvpd.o
../scripts/Makefile.build:361: recipe for target 'drivers/char/diag/diag_memorydevice.o' failed
  CC      drivers/clk/clk-divider.o
  CC      drivers/clocksource/arm_arch_timer.o
  CC      drivers/char/vservices_serial.o
../scripts/Makefile.build:676: recipe for target 'drivers/char/diag' failed
  CC      drivers/clk/clk-fixed-factor.o
  CC      drivers/clk/clk-fixed-rate.o
  CC      drivers/clocksource/dummy_timer.o
  CC      drivers/clk/clk-gate.o
  CC      drivers/clk/clk-multiplier.o
  CC      drivers/clk/clk-mux.o
  CC      drivers/clk/clk-composite.o
  AR      drivers/clocksource/built-in.o
  CC      drivers/clk/clk-fractional-divider.o
  CC      drivers/clk/clk-gpio.o
  CC      drivers/clk/clk-conf.o
  CC      drivers/clk/clk-xgene.o
../scripts/Makefile.build:676: recipe for target 'drivers/char' failed
  AR      drivers/clk/bcm/built-in.o
  AR      drivers/clk/imgtec/built-in.o
  AR      drivers/clk/mvebu/built-in.o
  AR      drivers/clk/renesas/built-in.o
  CC      drivers/clk/qcom/common.o
  AR      drivers/clk/ti/built-in.o
  CC      drivers/clk/qcom/clk-regmap.o
  CC      drivers/clk/qcom/clk-alpha-pll.o
  CC      drivers/clk/qcom/clk-pll.o
  CC      drivers/clk/qcom/clk-rcg.o
  CC      drivers/clk/qcom/clk-rcg2.o
  CC      drivers/clk/qcom/clk-branch.o
  CC      drivers/clk/qcom/clk-regmap-divider.o
  CC      drivers/clk/qcom/clk-regmap-mux.o
  CC      drivers/clk/qcom/clk-regmap-mux-div.o
  CC      drivers/clk/qcom/reset.o
  CC      drivers/clk/qcom/clk-voter.o
  CC      drivers/clk/qcom/clk-dummy.o
  CC      drivers/clk/qcom/clk-debug.o
  CC      drivers/clk/qcom/gdsc-regulator.o
  CC      drivers/clk/qcom/clk-cpu-osm.o
  CC      drivers/clk/qcom/camcc-sm8150.o
../scripts/Makefile.build:361: recipe for target 'drivers/clk/qcom/clk-alpha-pll.o' failed
../scripts/Makefile.build:676: recipe for target 'drivers/clk/qcom' failed
../scripts/Makefile.build:676: recipe for target 'drivers/clk' failed
/home/redbear/Xiaomi_Kernel_OpenSource-cepheus-q-oss/Makefile:1124: recipe for target 'drivers' failed
Makefile:146: recipe for target 'sub-make' failed
Makefile:24: recipe for target '__sub-make' failed

你们放在github上的源码是不是和推送给用户的内核的源码不一样啊？

YumeMichi, 2020-03-06:
麻烦放日志就放完整

MCredbear, 2020-03-06:
GEN     ./Makefile
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  GEN     ./Makefile
  WRAP    arch/arm64/include/generated/uapi/asm/errno.h
  WRAP    arch/arm64/include/generated/uapi/asm/ioctl.h
  WRAP    arch/arm64/include/generated/uapi/asm/ioctls.h
  WRAP    arch/arm64/include/generated/uapi/asm/ipcbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/kvm_para.h
  WRAP    arch/arm64/include/generated/uapi/asm/mman.h
  WRAP    arch/arm64/include/generated/uapi/asm/msgbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/poll.h
  WRAP    arch/arm64/include/generated/uapi/asm/resource.h
  WRAP    arch/arm64/include/generated/uapi/asm/sembuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/shmbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/socket.h
  WRAP    arch/arm64/include/generated/uapi/asm/sockios.h
  WRAP    arch/arm64/include/generated/uapi/asm/swab.h
  WRAP    arch/arm64/include/generated/uapi/asm/termbits.h
  WRAP    arch/arm64/include/generated/uapi/asm/termios.h
  WRAP    arch/arm64/include/generated/uapi/asm/types.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/basic/bin2c
  Using .. as source for kernel
  WRAP    arch/arm64/include/generated/asm/bugs.h
  WRAP    arch/arm64/include/generated/asm/clkdev.h
  WRAP    arch/arm64/include/generated/asm/delay.h
  WRAP    arch/arm64/include/generated/asm/div64.h
  WRAP    arch/arm64/include/generated/asm/dma.h
  WRAP    arch/arm64/include/generated/asm/dma-contiguous.h
  WRAP    arch/arm64/include/generated/asm/early_ioremap.h
  WRAP    arch/arm64/include/generated/asm/emergency-restart.h
  WRAP    arch/arm64/include/generated/asm/hw_irq.h
  WRAP    arch/arm64/include/generated/asm/irq_regs.h
  WRAP    arch/arm64/include/generated/asm/kdebug.h
  WRAP    arch/arm64/include/generated/asm/kmap_types.h
  WRAP    arch/arm64/include/generated/asm/local.h
  WRAP    arch/arm64/include/generated/asm/local64.h
  WRAP    arch/arm64/include/generated/asm/mcs_spinlock.h
  WRAP    arch/arm64/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/arm64/include/generated/asm/msi.h
  WRAP    arch/arm64/include/generated/asm/preempt.h
  WRAP    arch/arm64/include/generated/asm/qrwlock.h
  WRAP    arch/arm64/include/generated/asm/rwsem.h
  WRAP    arch/arm64/include/generated/asm/segment.h
  WRAP    arch/arm64/include/generated/asm/serial.h
  WRAP    arch/arm64/include/generated/asm/set_memory.h
  WRAP    arch/arm64/include/generated/asm/sizes.h
  WRAP    arch/arm64/include/generated/asm/switch_to.h
  WRAP    arch/arm64/include/generated/asm/trace_clock.h
  WRAP    arch/arm64/include/generated/asm/unaligned.h
  WRAP    arch/arm64/include/generated/asm/user.h
  WRAP    arch/arm64/include/generated/asm/vga.h
  WRAP    arch/arm64/include/generated/asm/xor.h
  HOSTCC  scripts/dtc/dtc.o
  CC      kernel/bounds.s
  CHK     include/generated/timeconst.h
  UPD     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  UPD     include/generated/bounds.h
  CC      arch/arm64/kernel/asm-offsets.s
  HOSTCC  scripts/dtc/flattree.o
  HOSTCC  scripts/genksyms/genksyms.o
  SHIPPED scripts/genksyms/parse.tab.c
  SHIPPED scripts/genksyms/lex.lex.c
  SHIPPED scripts/genksyms/parse.tab.h
[... 1611 lines cut ...]
  CC      net/ipv4/netfilter/arptable_filter.o
  CC      net/netfilter/xt_owner.o
  CC      net/ipv4/netfilter/nf_dup_ipv4.o
  CC      net/netfilter/xt_pkttype.o
  AR      net/ipv4/netfilter/nf_conntrack_ipv4.o
  AR      net/ipv4/netfilter/nf_nat_ipv4.o
  CC      net/netfilter/xt_policy.o
  CC      net/netfilter/xt_quota.o
  CC      net/netfilter/xt_quota2.o
  CC      net/netfilter/xt_socket.o
  CC      net/netfilter/xt_state.o
  CC      net/netfilter/xt_statistic.o
  CC      net/netfilter/xt_string.o
  CC      net/netfilter/xt_time.o
  CC      net/netfilter/xt_u32.o
  AR      net/netfilter/netfilter.o
  AR      net/netfilter/nf_conntrack.o
  AR      net/netfilter/nf_conntrack_h323.o
  AR      net/netfilter/nf_nat.o
  AR      net/ipv4/netfilter/built-in.o
  AR      net/ipv4/built-in.o
  AR      net/netfilter/built-in.o
  AR      net/built-in.o
Makefile:146: recipe for target 'sub-make' failed
Makefile:24: recipe for target '__sub-make' failed

MCredbear, 2020-03-11:
> 麻烦放日志就放完整

大佬，我放完整了，帮我看下行吗？。

YumeMichi, 2020-03-11:
> > 麻烦放日志就放完整
> 
> 大佬，我放完整了，帮我看下行吗？。

操作流程发一下 这日志看起来也没有明确的错误原因 确定是按Wiki里的步骤操作的？

MCredbear, 2020-03-11:
> > > 麻烦放日志就放完整
> > 
> > 
> > 大佬，我放完整了，帮我看下行吗？。
> 
> 操作流程发一下 这日志看起来也没有明确的错误原因 确定是按Wiki里的步骤操作的？

我是这样操作的：
解压到/home/用户名/
cd 源码目录
新建目录output
make xxxxxxxxx_config O=output
cd output
make -j8 O=output

YumeMichi, 2020-03-11:
> > > > 麻烦放日志就放完整
> > > 
> > > 
> > > 大佬，我放完整了，帮我看下行吗？。
> > 
> > 
> > 操作流程发一下 这日志看起来也没有明确的错误原因 确定是按Wiki里的步骤操作的？
> 
> 我是这样操作的：
> 解压到/home/用户名/
> cd 源码目录
> 新建目录output
> make xxxxxxxxx_config O=output
> cd output
> make -j8 O=output

交叉编译器呢

MCredbear, 2020-03-11:
> > > > > 麻烦放日志就放完整
> > > > 
> > > > 
> > > > 大佬，我放完整了，帮我看下行吗？。
> > > 
> > > 
> > > 操作流程发一下 这日志看起来也没有明确的错误原因 确定是按Wiki里的步骤操作的？
> > 
> > 
> > 我是这样操作的：
> > 解压到/home/用户名/
> > cd 源码目录
> > 新建目录output
> > make xxxxxxxxx_config O=output
> > cd output
> > make -j8 O=output
> 
> 交叉编译器呢

redbear@redbear-Lenovo-XiaoXin-14API-2019:~$ aarch64-linux-android-gcc -v
Using built-in specs.
COLLECT_GCC=aarch64-linux-android-gcc
COLLECT_LTO_WRAPPER=/home/redbear/google_gcc/aarch64-linux-android-4.9/bin/../libexec/gcc/aarch64-linux-android/4.9/lto-wrapper
Target: aarch64-linux-android
Configured with: /usr/local/google/buildbot/src/android/gcc/toolchain/build/../gcc/gcc-4.9/configure --prefix=/tmp/8bf1fa4a69e1670b34d7855ef9cf731f --target=aarch64-linux-android --host=x86_64-linux-gnu --build=x86_64-linux-gnu --with-gnu-as --with-gnu-ld --enable-languages=c,c++ --with-gmp=/buildbot/tmp/build/toolchain/temp-install --with-mpfr=/buildbot/tmp/build/toolchain/temp-install --with-mpc=/buildbot/tmp/build/toolchain/temp-install --with-cloog=/buildbot/tmp/build/toolchain/temp-install --with-isl=/buildbot/tmp/build/toolchain/temp-install --with-ppl=/buildbot/tmp/build/toolchain/temp-install --disable-ppl-version-check --disable-cloog-version-check --disable-isl-version-check --enable-cloog-backend=isl --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --disable-libssp --enable-threads --disable-nls --disable-libmudflap --disable-libgomp --disable-libstdc__-v3 --disable-sjlj-exceptions --disable-shared --disable-tls --disable-libitm --enable-initfini-array --disable-nls --prefix=/tmp/8bf1fa4a69e1670b34d7855ef9cf731f --with-sysroot=/tmp/8bf1fa4a69e1670b34d7855ef9cf731f/sysroot --with-binutils-version=2.25 --with-mpfr-version=3.1.1 --with-mpc-version=1.0.1 --with-gmp-version=5.0.5 --with-gcc-version=4.9 --with-gdb-version=none --with-gxx-include-dir=/tmp/8bf1fa4a69e1670b34d7855ef9cf731f/include/c++/4.9 --with-bugurl=http://source.android.com/source/report-bugs.html --enable-languages=c,c++ --disable-bootstrap --enable-plugins --enable-libgomp --enable-gnu-indirect-function --disable-libsanitizer --enable-gold --enable-ld=default --enable-threads --enable-eh-frame-hdr-for-static --enable-fix-cortex-a53-835769 --enable-graphite=yes --with-isl-version=0.11.1 --with-cloog-version=0.18.0 --program-transform-name='s&^&aarch64-linux-android-&' --enable-gold
Thread model: posix
gcc version 4.9 20150123 (prerelease) (GCC) 
在/etc/profile底部添加：
export ANDROID_AARCH64=/home/redbear/google_gcc/aarch64-linux-android-4.9/bin/
export ARCH=arm64
export SUBARCH=arm64
export PATH=$PATH:$ANDROID_AARCH64
export CROSS_COMPILE=aarch64-linux-android-

shandongtlb, 2020-03-11:
没用clang吗，只用gcc？

MCredbear, 2020-03-11:
> 
> 
> 没用clang吗，只用gcc？

是的

shandongtlb, 2020-03-11:
wiki上已经说了要用高通的clang

MCredbear, 2020-03-11:
> 
> 
> wiki上已经说了要用高通的clang

好吧，知道了，谢谢

Gonglja, 2020-03-11:
我也在编这个，目前也有一些问题，明天在接着看吧

MCredbear, 2020-03-14:
> wiki上已经说了要用高通的clang

我按照wiki上的做法用了高通的clang，还是失败了：

redbear@redbear-Lenovo-XiaoXin-14API-2019:~/Xiaomi_Kernel_OpenSource-cepheus-q-oss$ mkdir out
redbear@redbear-Lenovo-XiaoXin-14API-2019:~/Xiaomi_Kernel_OpenSource-cepheus-q-oss$ make O=out REAL_CC=/home/redbear/gaotong_toolchains/toolchains/llvm-Snapdragon_LLVM_for_Android_6.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu- raphael_user_defconfig
make[1]: 进入目录“/home/redbear/Xiaomi_Kernel_OpenSource-cepheus-q-oss/out”
  HOSTCC  scripts/basic/fixdep
  GEN     ./Makefile
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
#
# configuration written to .config
#
make[1]: 离开目录“/home/redbear/Xiaomi_Kernel_OpenSource-cepheus-q-oss/out”
redbear@redbear-Lenovo-XiaoXin-14API-2019:~/Xiaomi_Kernel_OpenSource-cepheus-q-oss$ make -j8 O=out REAL_CC=/home/redbear/gaotong_toolchains/toolchains/llvm-Snapdragon_LLVM_for_Android_6.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu- 2>&1 | tee kernel.log
make[1]: 进入目录“/home/redbear/Xiaomi_Kernel_OpenSource-cepheus-q-oss/out”
  GEN     ./Makefile
scripts/kconfig/conf  --silentoldconfig Kconfig
arch/arm64/Makefile:57: Detected assembler with broken .inst; disassembly will be unreliable
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  GEN     ./Makefile
  WRAP    arch/arm64/include/generated/uapi/asm/errno.h
  WRAP    arch/arm64/include/generated/uapi/asm/ioctls.h
  WRAP    arch/arm64/include/generated/uapi/asm/ioctl.h
  WRAP    arch/arm64/include/generated/uapi/asm/kvm_para.h
  WRAP    arch/arm64/include/generated/uapi/asm/ipcbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/mman.h
  WRAP    arch/arm64/include/generated/uapi/asm/msgbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/poll.h
  WRAP    arch/arm64/include/generated/uapi/asm/resource.h
  WRAP    arch/arm64/include/generated/uapi/asm/sembuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/shmbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/sockios.h
  WRAP    arch/arm64/include/generated/uapi/asm/socket.h
  WRAP    arch/arm64/include/generated/uapi/asm/swab.h
  WRAP    arch/arm64/include/generated/uapi/asm/termbits.h
  WRAP    arch/arm64/include/generated/uapi/asm/termios.h
  WRAP    arch/arm64/include/generated/uapi/asm/types.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/basic/bin2c
  Using .. as source for kernel
  WRAP    arch/arm64/include/generated/asm/bugs.h
  WRAP    arch/arm64/include/generated/asm/clkdev.h
  WRAP    arch/arm64/include/generated/asm/delay.h
  WRAP    arch/arm64/include/generated/asm/div64.h
  WRAP    arch/arm64/include/generated/asm/dma.h
  WRAP    arch/arm64/include/generated/asm/dma-contiguous.h
  WRAP    arch/arm64/include/generated/asm/early_ioremap.h
  WRAP    arch/arm64/include/generated/asm/emergency-restart.h
  WRAP    arch/arm64/include/generated/asm/hw_irq.h
  WRAP    arch/arm64/include/generated/asm/irq_regs.h
  WRAP    arch/arm64/include/generated/asm/kdebug.h
  WRAP    arch/arm64/include/generated/asm/kmap_types.h
  WRAP    arch/arm64/include/generated/asm/local.h
  WRAP    arch/arm64/include/generated/asm/local64.h
  WRAP    arch/arm64/include/generated/asm/mcs_spinlock.h
  WRAP    arch/arm64/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/arm64/include/generated/asm/msi.h
  WRAP    arch/arm64/include/generated/asm/preempt.h
  WRAP    arch/arm64/include/generated/asm/qrwlock.h
  WRAP    arch/arm64/include/generated/asm/rwsem.h
  WRAP    arch/arm64/include/generated/asm/segment.h
  WRAP    arch/arm64/include/generated/asm/serial.h
  WRAP    arch/arm64/include/generated/asm/set_memory.h
[... 3906 lines cut ...]
  AR      drivers/usb/gadget/function/usb_f_rndis.o
  AR      drivers/usb/gadget/function/usb_f_mass_storage.o
  AR      drivers/usb/gadget/function/usb_f_fs.o
  AR      drivers/usb/gadget/function/usb_f_uac2.o
  AR      drivers/usb/gadget/function/usb_f_midi.o
  AR      drivers/usb/gadget/function/usb_f_hid.o
  AR      drivers/usb/gadget/function/usb_f_ptp.o
  AR      drivers/usb/gadget/function/usb_f_audio_source.o
  AR      drivers/usb/gadget/function/usb_f_diag.o
  AR      drivers/usb/gadget/function/usb_f_mtp.o
  AR      drivers/usb/gadget/function/usb_f_ccid.o
  AR      drivers/usb/gadget/function/usb_f_accessory.o
  AR      drivers/usb/gadget/function/usb_f_cdev.o
  AR      drivers/usb/gadget/function/usb_f_qdss.o
  AR      drivers/usb/gadget/function/usb_f_gsi.o
  AR      drivers/usb/gadget/function/built-in.o
  AR      drivers/usb/gadget/built-in.o
  AR      drivers/usb/built-in.o
  AR      drivers/built-in.o
make[1]: 离开目录“/home/redbear/Xiaomi_Kernel_OpenSource-cepheus-q-oss/out”
Makefile:146: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2

@YumeMichi 再帮我看下？
还有就是wiki上的那个操作我属实没看懂，那个高通的clang里面是没有aarch64-linux-android-这类东西的，且环境变量里aarch64-linux-android-指向的依旧是google的gcc4.9，但是它make的参数里有个叫REAL_CC的参数，指向的是高通的clang

YumeMichi, 2020-03-14:
你没装dtc

shandongtlb, 2020-03-14:
https://github.com/shandongtlb/dtc-1.6.0

MCredbear, 2020-03-16:
> 你没装dtc
> 
> ```
> DTC arch/arm64/boot/dts/qcom/apq8016-sbc.dtb
> /bin/sh: 1: dtc: not found
> scripts/Makefile.lib:325: recipe for target 'arch/arm64/boot/dts/qcom/apq8016-sbc.dtb' failed
> ```
> 
> Wiki里编译的是P的内核 你自己换成Q的
> 
> ```
> For msm-4.14 like Mi 9
> 
> dtc must be from aosp source code（pie-release） Clang from qcom
> ```

我现在安装了apt源里的device tree compiler，还是不行，好像三源码问题（日志已经超过最大字数上限了，我只能传文件了）：

[kernel.log](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/files/4338144/kernel.log)

TheGlitchh, 2020-05-15:
i am having the same issue

nakixii, 2020-08-15:
放弃编译设备树，使用boot里提取出来的。

MCredbear, 2020-08-16:
> 放弃编译设备树，使用boot里提取出来的。

那么怎么让它不编译设备树呢？该注释掉什么还是直接make -i？还是make Image？

nakixii, 2020-08-16:
> > 放弃编译设备树，使用boot里提取出来的。
> 
> 那么怎么让它不编译设备树呢？该注释掉什么还是直接make -i？还是make Image？

注释掉arch/arm64/Makefile和arch/arm64/boot/Makefile里关于dts的行

GeekerLYP, 2020-12-06:
> wiki上已经说了要用高通的clang

你好，请问可以发一下wiki的网址嘛，谢谢啦

MCredbear, 2020-12-06:
> > wiki上已经说了要用高通的clang
> 
> 你好，请问可以发一下wiki的网址嘛，谢谢啦

https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki

GeekerLYP, 2020-12-06:
> > > wiki上已经说了要用高通的clang
> > 
> > 
> > 你好，请问可以发一下wiki的网址嘛，谢谢啦
> 
> https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki

谢谢:)

1848639547, 2024-07-26:
> 我按照wiki上的做法用了高通的clang，还是失败了：
>
> redbear@redbear-Lenovo-XiaoXin-14API-2019:~/Xiaomi_Kernel_OpenSource-cepheus-q-oss$ mkdir out [...]

请问你解决了这错误吗，我在编译pad6内核时出现了相同错误
make[3]: *** 没有规则可制作目标“arch/arm64/boot/dts/qcom/apq8016-sbc.dtb”，由“__build” 需求。 停止。
make[3]: *** 正在等待未完成的任务....
make[2]: *** [../scripts/Makefile.build:642：arch/arm64/boot/dts/qcom] 错误 2
make[1]: *** [arch/arm64/Makefile:172：dtbs] 错误 2
make[1]: *** 正在等待未完成的任务....