MiCode/Xiaomi_Kernel_OpenSource #1205: 请问：想要有root权限且内核可刷，该如何做？

wangcong15, 2020-03-07:
您好，
我有一台红米4A手机，开发版有root权限，希望在这个基础上可以刷自己编译的内核。我有看到咱们这个repo里面有对应型号的内核源码，也在issue中找到了对应的config，成功编译了内核。
可是咱们这边现在开源的是稳定版的内核吗？我尝试对boot.img解包、更换kernel image、再打包。然后刷进去以后手机无法正常开机。

请问：想要有root权限且内核可刷，该如何做？
谢谢

MCredbear, 2020-03-11:
？是开发版的吧，我的Mi  note pro系统是开发版的，刷这里的源码编译出的内核是可以开机的

wangcong15, 2020-03-11:
> ？是开发版的吧，我的Mi note pro系统是开发版的，刷这里的源码编译出的内核是可以开机的

谢谢您的回答。
请问是怎么刷的内核，是直接在fastboot刷boot.img，还是重新打包rom双清刷入的呢？
如果是fastboot直接刷的boot.img的话，解包之前的boot.img是取自手机，还是取自官方卡刷包啊？

我试了一下上面的几种办法，在我的手机上都不太行。。

MCredbear, 2020-03-11:
> > ？是开发版的吧，我的Mi note pro系统是开发版的，刷这里的源码编译出的内核是可以开机的
> 
> 谢谢您的回答。
> 请问是怎么刷的内核，是直接在fastboot刷boot.img，还是重新打包rom双清刷入的呢？
> 如果是fastboot直接刷的boot.img的话，解包之前的boot.img是取自手机，还是取自官方卡刷包啊？
> 
> 我试了一下上面的几种办法，在我的手机上都不太行。。

rom包里的内核我不知道为什么我这里解不了包，我是用镧工具箱（分区管理功能）把手机的内核导出来解的包，刷的时候也是用镧工具箱刷进去的（和fastboot刷boot.img应该是等效的）

wangcong15, 2020-03-11:
> > > ？是开发版的吧，我的Mi note pro系统是开发版的，刷这里的源码编译出的内核是可以开机的
> > 
> > 
> > 谢谢您的回答。
> > 请问是怎么刷的内核，是直接在fastboot刷boot.img，还是重新打包rom双清刷入的呢？
> > 如果是fastboot直接刷的boot.img的话，解包之前的boot.img是取自手机，还是取自官方卡刷包啊？
> > 我试了一下上面的几种办法，在我的手机上都不太行。。
> 
> rom包里的内核我不知道为什么我这里解不了包，我是用镧工具箱（分区管理功能）把手机的内核导出来解的包，刷的时候也是用镧工具箱刷进去的（和fastboot刷boot.img应该是等效的）

谢谢，我再试一下

YumeMichi, 2020-03-11:
用AnyKernel 不用麻烦的手动修改boot.img
https://github.com/osm0sis/AnyKernel3

wangcong15, 2020-03-11:
还是搞不定。。如果你们成功刷了kernel的话，麻烦帮忙看一下手机里上显示的内核版本，与对应branch项目里的版本一致吗？我手机上是3.18.24，开源的kernel是3.18.31 
:(

MCredbear, 2020-03-11:
> 
> 
> 还是搞不定。。如果你们成功刷了kernel的话，麻烦帮忙看一下手机里上显示的内核版本，与对应branch项目里的版本一致吗？我手机上是3.18.24，开源的kernel是3.18.31
> :(

我的Mi note pro是一致的

wangcong15, 2020-03-11:
> > 还是搞不定。。如果你们成功刷了kernel的话，麻烦帮忙看一下手机里上显示的内核版本，与对应branch项目里的版本一致吗？我手机上是3.18.24，开源的kernel是3.18.31
> > :(
> 
> 我的Mi note pro是一致的

谢谢您的回复：）

phone-burner, 2020-03-29:
> > 还是搞不定。。如果你们成功刷了kernel的话，麻烦帮忙看一下手机里上显示的内核版本，与对应branch项目里的版本一致吗？我手机上是3.18.24，开源的kernel是3.18.31
> > :(
> 
> 我的Mi note pro是一致的

你好，我的手机是米6X(wayne)，系统MIUI11 Android 9，内核4.4.153。用官方的源码和config编译以后启动不了，怎么解决？