MiCode/Xiaomi_Kernel_OpenSource #1207: [daisy-q-oss] black screen firmware!

Aarqw12, 2020-03-09:
hi ! 

After upgrade my daisy to android 10 my devices get black screen at all after splash , cant go to recovery ...
Before this upgrade I have compiled kernel xiaomi daisy-q-oss and get this same issue... 
Its over hard to give proper source kernel Q daisy?
You can see dmesg error related to kernel ...
thanks for your response ....

[dmesg(1).txt](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/files/4306494/dmesg.1.txt)

Aarqw12, 2020-03-10:
found for fix issue need change dtsi ... https://github.com/Aarqw12/kernel_xiaomi_daisy-1/commit/367af53b5ee41d02e4061eed6644db4ff8995a10

Aarqw12, 2020-03-16:
- found another issue :-1: 

-  kernel can't see cpu temperature !

phone-burner, 2020-03-28:
Hello, could you tell me how to compile a bootable kernel ?
 My device is MI 6X (Wayne) with MIUI 11 20.3.26 . I downloaded the source code for Wayne Android 9 version 4.4.153 . I ran these command in the chroot environment ( Ubuntu 20.04 ARM64 ) on my phone . 
make wayne-perf_defconfig
make menuconfig
make CC=gcc-8 -j8
I enabled KVM feature & NTFS support in the config .
After the process finished, I extracted the original boot.img with command 'abootimg -x boot.img' , and then generated a new one with command 'mkbootimg --kernel Image.gz-dtb --ramdisk initrd.img --cmdline bootimg.cfg -o new_boot.img' using the new Image.gz-dtb .
I signed the new_boot.img with my key by the method on xda.
After flashing the boot image, the phone stuck on the first screen showing 'unlocked' . 
What's the matter with my kernel ?

Aarqw12, 2020-04-02:
why you no compile pure kernel source with only wifi added in 1st time for see?
and better to make your kernel into zip for flash in recovery its more simple and speed
> Hello, could you tell me how to compile a bootable kernel ?
> My device is MI 6X (Wayne) with MIUI 11 20.3.26 . I downloaded the source code for Wayne Android 9 version 4.4.153 . I ran these command [...]

Aarqw12, 2020-04-12:
1 more issue found in kernel daisy-q-oss change in  drivers/leds/leds-qpnp-flash.c have missing line after " turn_off: "

result for user : in offline charging leds no work !

fix? found line missing in daisy-p-oss change ...