MiCode/Xiaomi_Kernel_OpenSource #1208: daisy-q-oss: broked bootable on some hardware parts

eremitein, 2020-03-14:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/daisy-q-oss

This kernel have major issues, which caused briking devices with CSOT + fts_ts hardware parts in February Android One 10 update

Pay attention on this derp, guys

eremitein, 2020-03-14:
Also check problems with screen lightning (broked at bottom section) and led indicator (too dim)

[image]

pawelik001, 2020-03-14:
Hello, as someone said on Mi A2 Lite telegram group, screen lighting (as @eremitein said) is caused by providing too much voltage from kernel said, and this is not temporary, but permanent damage. 
Cheers :)

Aarqw12, 2020-03-18:
+found thermal change cpu is brocken! also change camera_v2 is very light in my opinions in sources ( not surprised for gcam bug video in default gcam setting )

> Hello, as someone said on Mi A2 Lite telegram group, screen lighting (as @eremitein said) is caused by providing too much voltage from kernel said, and this is not temporary, but permanent damage.
> Cheers :)

Aarqw12, 2020-04-12:
1 more issue found in kernel daisy-q-oss change in drivers/leds/leds-qpnp-flash.c have missing line after " turn_off: "

result for user : in offline charging leds no work !

fix? found line missing in daisy-p-oss change ...