MiCode/Xiaomi_Kernel_OpenSource #1210: Redmi 7/Y3 Android 10

mylove90, 2020-03-16:
Dear Xiaomi,

Good day.

It has been long since Miui 11 released for our device onc/onclite but with Android 9.0....Will the device receive an update to Android 10 or not ????

We already got multiple custom roms for Android 10...So we are sure our device can handle it very well.

Thanks in advance
Best regards

mylove90, 2020-03-30:
WOW....Xiaomi devs are so professionals that they didn't even look at that open issue !

Loco1911, 2020-06-04:
Update Redmi 7 to Android 10!!!

thatisfolks302, 2020-06-04:
UPDATE XIAXAXAXAXAXAXXIOMI REDMEME 7 TO ANDROID 10 OR I WILL CRI EVERTM

thatisfolks302, 2020-06-04:
THIS IS SERIUSLY IM NOT JOKEN SRY FOR BAD ENGLAND UWU

Renshi15, 2020-06-04:
We love our Redmi 7 onclite, pls make an update for android 10

Zelectic, 2020-06-04:
GIVE US THE UPDATE WTF? ANDROID 10 CMON

junior0liveira, 2020-06-05:
Xiaomi where is the update for android 10 on Redmi 7 onclite !?

TheAcanthite, 2020-06-05:
Gib updoot for Xiaomi Redmi 7 phassssssttttttttttttt

Risti699, 2020-06-05:
Redmeme 7 android 10 when plxx

BitesFor, 2020-06-06:
Dear Xiaomi, Please start working on Android 10, which you promised us. Don't forget us !!

asnoman2, 2020-06-06:
Please Android 10 with miui 12

ragy365, 2020-06-08:
Gib Redmi 7 Android 10 update , plox

kairusds, 2020-06-13:
up

PIPIPIG233666, 2020-06-13:
Down
This is not the right place...

ragy365, 2020-06-13:
> Down
> This is not the right place...

Up
It is

hamohamo6, 2020-06-14:
Can we get android 10 already

Renshi15, 2020-06-15:
Pls release android 10 for our redmi 7

TheAcanthite, 2020-06-28:
Android 10 China released. When releasing sources?!

mylove90, 2020-08-08:
Thanks for releasing the sources :)

closed 2020-08-08 by mylove90