MiCode/Xiaomi_Kernel_OpenSource #1213: Please repair MI9/M9Tpro's Android10 fingerprint driver as soon as possible, thanks// 请尽快修复米9和k20p 安卓10的指纹驱动，谢谢😃

shandongtlb, 2020-03-21:
尊敬的开发者，因为最近一系列原因知道你们最近很忙，请问米9k20p 安卓10的指纹驱动什么时候能修好啊，换句话说，MIUI稳定版什么时候发布呀

Dear developers, I know you are very busy. When can I fix the fingerprint driver of m9k20p Android 10? In other words, when will the stable version of MIUI be released.

開発者の皆様、最近はさまざまな理由で忙しいのですが、Mi 9k20p Android 10の指紋ドライバーを修復できる時期、つまりMIUIの安定バージョンはいつリリースされるのでしょうか？

Drodzy deweloperzy, wiem, że ostatnio byłeś zajęty z kilku powodów: czy mogę zapytać, kiedy sterownik odcisków palców Mi 9k20p dla Androida 10 można naprawić, innymi słowy, kiedy zostanie wydana stabilna wersja MIUI?

친애하는 개발자 여러분, 최근 몇 가지 이유로 바쁘신 것으로 알고 있습니다. Mi 9k20p Android 10의 지문 드라이버를 언제 수리 할 수 ​​있는지, 즉 MIUI 안정 버전이 언제 출시 될지 묻습니다.

Sehr geehrter Entwickler, ich weiß, dass Sie in letzter Zeit aus einer Reihe von Gründen beschäftigt waren. Darf ich fragen, wann der Fingerabdrucktreiber von Mi 9k20p Android 10 repariert werden kann, mit anderen Worten, wann die stabile MIUI-Version veröffentlicht wird?

Estimado desarrollador, sé que ha estado ocupado recientemente por una serie de razones: ¿Puedo preguntar cuándo se puede reparar el controlador de huellas digitales de Mi 9k20p Android 10, en otras palabras, cuándo se lanzará la versión estable de MIUI?

Уважаемый разработчик, я знаю, что вы были заняты в последнее время по ряду причин. Могу ли я спросить, когда можно восстановить драйвер отпечатка пальца Mi 9k20p Android 10, другими словами, когда будет выпущена стабильная версия MIUI?

Cher développeur, je sais que vous avez été occupé récemment pour une série de raisons. Puis-je demander quand le pilote d'empreinte digitale de Mi 9k20p Android 10 peut être réparé, en d'autres termes, quand la version stable de MIUI sera-t-elle publiée

מפתח יקר, אני יודע שעסקת לאחרונה מסדרת סיבות, האם מותר לי לשאול מתי ניתן לתקן את מנהל ההתקן של טביעות האצבע של Mi 9k20p אנדרואיד 10, או במילים אחרות, מתי תשוחרר גרסת MIUI יציבה?

Caro sviluppatore, so che sei stato occupato di recente per una serie di motivi. Posso chiederti quando il driver dell'impronta digitale di Mi 9k20p Android 10 può essere riparato, in altre parole, quando verrà rilasciata la versione stabile MIUI

谢谢XD

EviraKernel, 2020-03-25:
we wait miui answer!!

shandongtlb, 2020-03-26:
> we wait miui answer!!

Yes!!

phone-burner, 2020-03-28:
Yes!!!

StorySea, 2020-04-15:
no answer .
[image]

shandongtlb, 2020-04-15:
> no answer .
> [image]

hiahiahia

EviraKernel, 2020-04-29:
so miui how time we need wait for relase source with fp fixed? beta relase of miui 12 was relased i think on june relase stable miui 12 kernel source is the same so why not relase source?

shandongtlb, 2020-04-30:
> so miui how time we need wait for relase source with fp fixed? beta relase of miui 12 was relased i think on june relase stable miui 12 kernel source is the same so why not relase source?

I think FP will be released in June, or has given up