MiCode/Xiaomi_Kernel_OpenSource #1215: Mi mix 3 5g - Android ten kernel 4.14.117+ or Device tree

ShqipLewis, 2020-03-30:
[image]

Well i see there is on going work behind the scenes for our loved mi mix 3 5g andromeda, the latest being 20.3.6 , it would be nice if you guys could collaborate with us here and provide us the device trees/vendor/kernel for the miui 11 and q of this device, thank you :)

TheMalachite, 2020-04-09:
> [image]
> 
> Well i see there is on going work behind the scenes for our loved mi mix 3 5g andromeda, the latest being 20.3.6 , it would be nice if you guys could collaborate with us here and provide us the device trees/vendor/kernel for the miui 11 and q of this device, thank you :)

Xiaomi will never release their device and vendor tree for MIUI to us so forget about this, and for kernel seems they stoped to roll out kernel update actually, due to the corona virus for sure

wmlwm, 2020-04-21:
**_Dear Xiaomi, we need Kernel Open Source android Q for MI MIX 3 5G. Do not ignore your users_**