MiCode/Xiaomi_Kernel_OpenSource #1217: No se puede habilitar KVM en el kernel de Linux a través de la edición de archivos de configuración.

tomchrisdperez, 2020-04-01:
Como se muestra arriba, mi teléfono es xiaomi 5 (msm8996). Edité "gemini_defconfig" y agregué las siguientes cuatro líneas:

CONFIG_VIRTUALIZATION=y 
CONFIG_KVM=y
CONFIG_KVM_MMIO=y 
CONFIG_KVM_ARM_HOST=y

Sin embargo, cuando lo compilé y lo actualicé en MIUI 10, el archivo "/ dev / kvm" que muestra si ha habilitado KVM con éxito en su teléfono no existe. Por lo tanto, no puedo usar la aceleración kvm cuando uso QEMU para emular un sistema de arquitectura nativo.
¿Cómo abordar el problema?

phone-burner, 2020-04-06:
I have not tested my KVM kernel yet , but I think using 'make menuconfig' to change the config may be better .

tomchrisdperez, 2020-04-07:
But menuconfig will reconfigure neccessary options in original configuration file "gemini_config".

tomchrisdperez, 2020-04-19:
Veo. No podemos tomar el control de EL2 en modo de arranque inseguro. EL2 es esencial para el hipervisor que puede crear una máquina virtual en un sistema operativo en ejecución simultáneamente.

closed 2020-04-19 by tomchrisdperez