MiCode/Xiaomi_Kernel_OpenSource #1228: Kernel source for Android 10, sdm710 devices

TheMalachite, 2020-04-20:
When are you gonna release those damn kernel source ? sirius, grus, pyxis and vela users have enough to wait !

That's been already more 3 month since Android 10 MIUI 11 stable update was released for sirius and grus and mostly 2 month for pyxis and vela !

begonia users got android 10 MIUI 11 stable since 1 month and they have android 10 kernel source !

We want sdm710 android 10 kernel sources !

@ivan19871002 @mi-code

EduardoSantinDev, 2020-04-20:
Where's the kernel source for pyxis and vela? We've been waiting for long enough! 
We want sdm710 android 10 kernel sources!
@mi-code

WiZaRd981, 2020-04-20:
> When are you gonna release those damn kernel source ? sirius, grus, pyxis and vela users have enough to wait !
>
> That's been already more 3 month since Android 10 MIUI 11 stable update was released [...]

Sadly how we need to create issue to get through Xiaomi.. Unbelievable.

czarsuperstar, 2020-04-20:
Please release the kernel source so we can maintain our device properly. Thank you Xiaomi

mutlucakar, 2020-04-20:
Please release the kernel source so we can maintain our device properly. Thank you Xiaomi

mutlucakar, 2020-04-20:
Where's the kernel source for pyxis and vela? We've been waiting for long enough!
We want sdm710 android 10 kernel sources!
@mi-code

JaviLerma, 2020-04-20:
Release kernel source!

nb-lunzz, 2020-04-20:
Release kernel source!!!

mizzerx, 2020-04-20:
Release Kernel Source

Maxxtrof, 2020-04-20:
Xiaomi, you better release the source!

qttc, 2020-04-20:
release kernel sources for sdm710 device!

isnthzy, 2020-04-20:
Release kernel source!!!

pokohontos, 2020-04-20:
Release kernel source!

SergeNu, 2020-04-20:
We want sdm710 android 10 kernel sources for maintaining Mi8 SE! Please!

alanmk24, 2020-04-20:
We need the kernel, kernel, kernel,,,sir Xiaomi.there is a silly Bluetooth bug and its even on your fastboot global rom,if you release the kernel sources maybe someone could help solve this problem.they say change is inevitable Xiaomi, go with it,,, RELEASE THE HOUNDS SMITHERS..NO, I MEAN KERNEL, XIAOMI -MASTERS.Release the source

vhdsz, 2020-04-20:
Release kernel source!!!

Punisher097, 2020-04-20:
Release Kernel source for Pyxis !

zaiserr, 2020-04-20:
Pls release Android 10 kernel for any sdm710 devices (That's been already 3 month since update are stable for sdm710 Xiaomi devices) !

Redmi Note 8 Pro users got Android 10 stable update 1 month ago and they got Android 10 kernel sources 4 day ago !

NaassomDourado, 2020-04-20:
Release Android 10 kernel for any sdm710 devices.

cok3307788, 2020-04-20:
> When are you gonna release those damn kernel source ? sirius, grus, pyxis and vela users have enough to wait !
>
> That's been already more 3 month since Android 10 MIUI 11 stable update was released [...]

Are you serious?
when you try to build Android 10 kernel sources code of Begonia , you will got lots of errors that the compilation process has to be terminated.

vd1719, 2020-04-20:
Please release Android 10 kernel source for sdm710 devices

imansh50, 2020-04-20:
Please release Android 10 kernel source for sdm710 devices

ha3ansol, 2020-04-20:
Please release Android 10 kernel source for sdm710 devices

maziix72, 2020-04-20:
Please release Android 10 kernel source for sdm710 devices

p6tr, 2020-04-20:
Please release Android 10 kernel source for sdm710 devices

ghasem7704, 2020-04-20:
Please release Android 10 kernel source for sdm710 devices

luk1337, 2020-04-20:
Please don't release Android 10 kernel source for sdm710 devices

atomsand, 2020-04-21:
> Please don't release Android 10 kernel source for sdm710 devices

lol

raphaeell, 2020-04-21:
Please release Android 10 kernel source for sdm710 devices! @mi-code

NaassomDourado, 2020-04-24:
> Please don't release Android 10 kernel source for sdm710 devices

you ghey kkkkkkkkkkkkk

NaassomDourado, 2020-04-24:
Please release Android 10 kernel source for sdm710 devices! @mi-code

NaassomDourado, 2020-04-25:
@mi-code release this.

maosilva, 2020-04-25:
Xiaomi Release the Pyxis kernel now! #ReleaseKernelPyxis #FredomForUsers

NaassomDourado, 2020-04-25:
Xiaomi Release the Pyxis kernel now! #ReleaseKernelPyxis #FreedomForUsers

NaassomDourado, 2020-04-25:
Pls release Android 10 kernel for any sdm710 devices (That's been already 3 month since update are stable for sdm710 Xiaomi devices) !

alexmadureira, 2020-04-25:
Xiaomi pls release Pyxis Kernel!!! 
#ReleaseKernelPyxis #FreedomForUsers

IsaccSaturno, 2020-04-25:
> When are you gonna release those damn kernel source ? sirius, grus, pyxis and vela users have enough to wait !
>
> That's been already more 3 month since Android 10 MIUI 11 stable update was released [...]

When are you going to release the damn pyxis kernel source? lack of respect with consumers, other brands make it available immediately after launch. Be careful not to lose more market for this #xiaomi

IsaccSaturno, 2020-04-25:
When are you going to release the damn pyxis kernel source? lack of respect with consumers, other brands make it available immediately after launch. Be careful not to lose more market for this #xiaomi

xDwinN, 2020-04-25:
Xiaomi, it's time to release this kernel for snap710 devices with Android 10. It's been over 3 months now and we urgently need the kernel. We can't wait any longer !!

wilson198918384, 2020-04-25:
Release kernel source!

ELFrathos, 2020-04-25:
@mi-code Where's the kernel source for pyxis and vela? We've been waiting for long enough!
We want sdm710 android 10 kernel sources!

alexandres-pyxis, 2020-04-25:
release kernel sources for sdm710 device!

mathlourenco94, 2020-04-25:
Xiaomi Release the Pyxis kernel now! #ReleaseKernelPyxis #FreedomForUsers

nullbytepl, 2020-04-25:
As if they care

granitzheg, 2020-04-26:
Xiaomi Release the Pyxis kernel now! #ReleaseKernelPyxis #FreedomForUsers

tauisilva, 2020-04-26:
Please release Android 10 kernel source for sdm710 devices

tauisilva, 2020-04-26:
Xiaomi Release the Pyxis kernel now! #ReleaseKernelPyxis #FreedomForUsers

pozhiloy-enotik, 2020-04-26:
Release kernel source!

francixcopp30, 2020-04-26:
release the Pyxis kernel, please

JimmyBondi, 2020-04-26:
time to release Pyxis kernel source, please

Luigi89FG, 2020-04-26:
release Android 10 kernel for any sdm710 devices!

alexamodei, 2020-04-26:
Please release android 10 kernel source for our smd710 devices

NaassomDourado, 2020-04-26:
Release pyxis kernel source @mi-code

ELFrathos, 2020-04-26:
Release pyxis kernel source @mi-code

bncassaro, 2020-04-26:
Release pyxis kernel source @mi-code

ELFrathos, 2020-04-26:
Release pyxis kernel source Snapdragon 710 and Android 10 @mi-code

pozhiloy-enotik, 2020-04-27:
Could you please release kernel source for sirius, grus, pyxis and vela @mi-code

bgcngm, 2020-04-27:
STOP!!!

lapinpt, 2020-04-28:
Please unleash pyxis smd710 kernel source

NaassomDourado, 2020-04-29:
Release @mi-code

DobreDomitian, 2020-04-30:
Release kernel source pls @ivan19871002 @mi-code

[3 more replies]

closed 2020-05-31 by TheMalachite