MiCode/Xiaomi_Kernel_OpenSource #1245: 请求更新小米9/小米9 Pro 5G/红米K20Pro的内核源码

Aquarius223, 2020-05-19:
我们需要在这些手机上进行AOSP的调试，使用了MIUI最新的Blob/firmware，以及最新的库存驱动，目前在2020年5月18日上指纹已经无法使用，按照小米承诺的稳定版更新即开源，那么在MIUI12稳定版发布之后能不能更新源码？  否则这将对于安卓开发者第一步先厌弃小米。

We need to debug AOSP on these phones, using the latest Blob / firmware of MIUI, and the latest inventory driver. The fingerprint is no longer available on May 18, 2020. According to the stable version update promised by Xiaomi, that is open source, Can the source code be updated after the stable version of MIUI12 is released? Otherwise, this will be the first step for Android developers to abandon Xiaomi.

shandongtlb, 2020-05-19:
+1，，话说安卓11都快好了

panqiang-code, 2020-05-20:
+1，不更新源码就没了可玩性，转战一加算了

18cmDeKun, 2020-06-16:
同求再不更新真的没有可玩性目前目前手持9Pro

SakuraTearsM, 2020-06-19:
我需要最快的更新

已从 Outlook Mobile<https://aka.ms/blhgte> 发送