MiCode/Xiaomi_Kernel_OpenSource #1246: Found more defconfig file in arch/arm64/config/ folder in redmi note 7

figforg, 2020-05-25:
I found many defconfig file in path arch/arm64/config folder.
I'not sure which one must be modify for building compile kernel of Redmi Note 7.
Please help me.

I found below defconfig file in config folder.

defconfig                 msm-auto-gvm-perf_defconfig        msm_defconfig
fsmcortex_defconfig       msm-auto-perf_defconfig            msm-perf_defconfig
fsmcortex-perf_defconfig  msmcortex_defconfig                ranchu64_defconfig
lavender-perf_defconfig   msmcortex_mediabox_defconfig       sdm660_defconfig
msm-auto_defconfig        msmcortex_mediabox-perf_defconfig  sdm660-perf_defconfig
msm-auto-gvm_defconfig    msmcortex-perf_defconfig

PIPIPIG233666, 2020-05-25:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/1b35cb9e684cbc58867ee44718eb92e7ff951b3a
`The kernel config file used is lavender-perf_defconfig`

figforg, 2020-05-26:
Yes, I compiled that lavender-perf_defcnfig file.
But I found below error
fatal error: btfm_slim.h: No such file or directory
 #include <btfm_slim.h>
                       ^
compilation terminated.
make[2]: *** [scripts/Makefile.build:283: drivers/bluetooth/btfm_slim.o] Error 1
make[1]: *** [scripts/Makefile.build:489: drivers/bluetooth] Error 2
make: *** [Makefile:1010: drivers] Error 2

How can I fix it?
I saw some fixed solution in forum as : replacing <> with "" in header file.
But I don't understand where is header file and where I should replace.
Please help me.

figforg, 2020-05-26:
> [1b35cb9](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/1b35cb9e684cbc58867ee44718eb92e7ff951b3a)
> `The kernel config file used is lavender-perf_defconfig`

Thank a lot.

Aquarius223, 2020-05-29:
将输出目录定向到out文件夹
即make O=out xxxxxx_defconfig
然后make O=out -j线程数即可

Direct the output directory to the out folder
Ie make O = out xxxxxx_defconfig
Then make O = out -j thread count