MiCode/Xiaomi_Kernel_OpenSource #1253: Source for Redmi Note 9 (merlin)

theGreatRajan, 2020-06-05:
Hi, dear Xiaomi guys, when will we have source for Redmi Note 9, we already have Redmi Note 9 Pro now.

zhihuiyuze, 2020-07-24:
我也需要红米9的内核，我想移植nethunter

zhihuiyuze, 2020-07-24:
Redmi 9

HemanthJabalpuri, 2020-12-10:
#1292, #1308, #1319, #1335, #1339, #1347

HemanthJabalpuri, 2020-12-22:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/lancelot-q-oss

lukezicamala, 2021-01-07:
> https://www.xda-developers.com/xiaomi-kernel-source-redmi-9-prime-redmi-note-9-4g-redmi-9-power-mi-10t-lite-redmi-note-9-pro-5g/
> 
> https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/lancelot-q-oss

 Hi friend, does this mean they finally realised it ? 
I have the redmi note 9 Merlin and im dying to install a custom rom 
Or is it for every other note 9/9 power or whatever...

HemanthJabalpuri, 2021-01-08:
> Hi friend, does this mean they finally realised it ?
I have the redmi note 9 Merlin and im dying to install a custom rom
Or is it for every other note 9/9 power or whatever...

We don't have custom rom for now. We can root, flash GSI and many more without TWRP and Kernel sources.
See https://forum.xda-developers.com/t/guides-faq-merlin-bootloader-magisk-edxposed-safetynet-gsi-treble-roms.4176581/