MiCode/Xiaomi_Kernel_OpenSource #1265: Source for Redmi Note 9 Pro Global (joyeuse)

chjohn22, 2020-06-10:
Could you add the kernel source code for the Redmi Note 9 Pro Global (joyeuse)?
https://www.gsmarena.com/xiaomi_redmi_note_9_pro-10217.php

spikespaz, 2020-06-11:
According to the research that I have done the kernel for the Redmi Note 9 Pro Max should be compatible with the Redmi Note 9 Pro Global (Joyeuse). Could someone from the Xiaomi team confirm whether or not the Curtana kernel is compatible with Joyeuse?

yudiwdynto, 2020-06-14:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/curtana-q-oss

yudiwdynto, 2020-06-14:
custa_* -> curtana
custd_* -> durandal
custe_* -> excalibur
custg_* -> gram
custj_* -> joyeuse

closed 2020-07-03 by chjohn22

1DeMaCr, 2020-07-24:
Not working on joyeuse, kernel can not start.

mapr4ve, 2021-04-11:
> custa_* -> curtana
> custd_* -> durandal
> custe_* -> excalibur
> custg_* -> gram
> custj_* -> joyeuse

how to specifically create it for Joyeuse? Is there any command?