MiCode/Xiaomi_Kernel_OpenSource #1267: Compile error for Mi Note 10

alibei, 2020-06-18:
Hi when i try to compile the kernel for tucana I get the following error:

  CC      techpack/audio/ipc/apr.o
  CC      techpack/audio/ipc/apr_v2.o
  CC      techpack/audio/ipc/apr_tal_rpmsg.o
  CC      techpack/audio/ipc/wcd-dsp-glink.o
  AR      techpack/audio/ipc/apr_dlkm.o
  AR      techpack/audio/ipc/wglink_dlkm.o
  AR      techpack/audio/ipc/built-in.o
  AR      techpack/audio/built-in.o
  AR      techpack/built-in.o
  GEN     .version
  CHK     include/generated/compile.h
  UPD     include/generated/compile.h
  CC      init/version.o
  AR      init/built-in.o
  AR      built-in.o
  LD      vmlinux.o
built-in.o: member techpack/audio/dsp/mius in archive is not an object
make[1]: *** [/home/alibei/android/lineage/kernel/xiaomi/android_kernel_xiaomi_tucana/Makefile:1119: vmlinux] Error 1
make[1]: Leaving directory '/home/alibei/android/lineage/out/target/product/tucana/obj/KERNEL_OBJ'
make: *** [Makefile:146: sub-make] Error 2
make: Leaving directory '/home/alibei/android/lineage/kernel/xiaomi/android_kernel_xiaomi_tucana'
16:44:56 ninja failed with: exit status 1

How can i fix that?

PIPIPIG233666, 2020-06-18:
Remove mius from makefile

closed 2020-06-18 by alibei

Snow-tear, 2020-07-18:
I have the same problem. what does "Remove mius from makefile" mean?

alibei, 2020-07-18:
In the teckpack folder there is a Makefile. You have to remove the lines from 33 - 38 and 63 - 67.