MiCode/Xiaomi_Kernel_OpenSource #1278: 请教一下为什么编译失败 I want to know why i failed in compiling.

Snow-tear, 2020-07-18:
你好, 我第一次编译内核, 一直没有成功. 我已经换了好多系统, 好多交叉编译器, 好多内核源码了, 在make的时候都会报错. 这里是kernel.log的输出文件, 烦请看一下出什么问题了. https://pastebin.com/aLghR8NM

Hello, this is my first time compiling Android kernel, but failed for many times. I have compiled it on different systems including Ubuntu and Debian. I also tried different cross compilers from ARM and google but all failed. Here is my kernel.log file. Could you please have a look at it and tell me where I was wrong?
https://pastebin.com/aLghR8NM

phone-burner, 2020-07-24:
换一个编译器？
用clang试试看
https://github.com/kdrag0n/proton-clang

Snow-tear, 2020-07-25:
谢谢楼上, 我的问题已经解决了, 我换了2016年的老编译器就成功了.

Snow-tear, 2020-07-25:
> 换一个编译器？
> 用clang试试看
> https://github.com/kdrag0n/proton-clang

谢谢楼上, 我的问题已经解决了, 我换了2016年的老编译器就成功了.

closed 2020-07-25 by Snow-tear