MiCode/Xiaomi_Kernel_OpenSource #1280: Can not boot joyeuse

1DeMaCr, 2020-07-24:
I built kernel for Redmi Note 9 Pro EU (joyeuse), but kernel can not start and have bootloop at first screen. last_kmsg is empty.
Any idea? What logs i can get?

PIPIPIG233666, 2020-07-24:
Flash dtbo

1DeMaCr, 2020-07-24:
> Flash dtbo

How to build dtbo.img?

PIPIPIG233666, 2020-07-24:
Get mkdtimg from Google

1DeMaCr, 2020-07-25:
> Get mkdtimg from Google

Can I use mkdtboimg.py from libufdt-master-utils ?
And what should i flash - Image.gz or Image.gz-dtb (using Anykernel3)?

1DeMaCr, 2020-07-27:
Tryed flash Image.gz/Image.gz-dtb + dtbo.img, Image.gz + dtb.img with and without dtbo.img, repack boot.img with Image.gz and dtb.img + dtbo.img. Can not boot.

hypeartist, 2020-08-04:
> Tryed flash Image.gz/Image.gz-dtb + dtbo.img, Image.gz + dtb.img with and without dtbo.img, repack boot.img with Image.gz and dtb.img + dtbo.img. Can not boot.

Same here. Any tips how to make it boot on MIUI11?

hypeartist, 2020-08-05:
> Tryed flash Image.gz/Image.gz-dtb + dtbo.img, Image.gz + dtb.img with and without dtbo.img, repack boot.img with Image.gz and dtb.img + dtbo.img. Can not boot.

I've managed to make it boot finally.

1DeMaCr, 2020-08-11:
> I've managed to make it boot finally.

How?

hypeartist, 2020-08-12:
> > I've managed to make it boot finally.
> 
> How?

Set path to AOSP dtc via DTC_EXT variable, then add this in ```arch/arm64/boot/dts/Makefile```:

```
	dtb-$(CONFIG_ARCH_ATOLL) += \
		custa-atoll-idp-overlay.dtb\
		custd-atoll-idp-overlay.dtb\
		custe-atoll-idp-overlay.dtb\
		custj-atoll-idp-overlay.dtb\
		custg-atoll-idp-overlay.dtb\
		atoll-idp-overlay.dtb\
		atoll-atp-overlay.dtb\
		atoll-qrd-overlay.dtb\
		atoll-wcd937x-idp-overlay.dtb\
		atoll-usbc-idp-overlay.dtb\
		atoll-rumi-overlay.dtb\
		atollp-idp-overlay.dtb\
		atollp-atp-overlay.dtb\
		atollp-qrd-overlay.dtb\
		custa-atoll-ab-idp-overlay.dtb\
		custd-atoll-ab-idp-overlay.dtb\
		custe-atoll-ab-idp-overlay.dtb\
		custj-atoll-ab-idp-overlay.dtb\
		custg-atoll-ab-idp-overlay.dtb\
		atoll-ab-idp-overlay.dtb\
		atoll-ab-atp-overlay.dtb\
		atoll-ab-qrd-overlay.dtb
```
inside of ```ifeq ($(CONFIG_BUILD_ARM64_DT_OVERLAY),y)``` condition branch
After all call ```mkdtboimg.py create dtbo.img dts/*/*.dtbo``` inside of ```out/arch/arm64/boot/```
That worked fine for me.