MiCode/Xiaomi_Kernel_OpenSource #1285: umi-q source

EviraKernel, 2020-08-04:
umi q source not boot i compiled all add tekpack qualcom wifi not boot maybe need fix by xiaomi

EviraKernel, 2020-08-12:
> umi q source not boot i compiled all add tekpack qualcom wifi not boot maybe need fix by xiaomi

ok now source boot but audio not work any solution?

hellobbn, 2020-08-13:
Hi, can you explain how to solve the "cannot booting" problem? I am stuck at the first boot screen (shows "mi" for a while and reboots).
What I have done:
- Added needed config (Build dtb and force loading kernel)
- Copy files in __kernel/device_tree__ to __arch/arm64/boot/dts__
- Cherry-pick the camera dtsi from your repository
- Compile the dtbo.img using `python mkdtboimg.py create dtbo.img ${src}/out/arch/arm64/boot/dts/qcom/*.dtbo`
- Replace the kernel in boot.img with `Image`