MiCode/Xiaomi_Kernel_OpenSource #1291: latest kernel source build failed

zhuang137, 2020-08-31:
android/xiaomi/cmi/kernel/arch/arm64/boot/dts/qcom/kona.dtsi:5149:35: fatal error: camera/kona-camera.dtsi: No such file or directory

xiangfeidexiaohuo, 2020-09-01:
> android/xiaomi/cmi/kernel/arch/arm64/boot/dts/qcom/kona.dtsi:5149:35: fatal error: camera/kona-camera.dtsi: No such file or directory

It's not like this, it's shielded or deleted.

zhuang137, 2020-09-02:
> > android/xiaomi/cmi/kernel/arch/arm64/boot/dts/qcom/kona.dtsi:5149:35: fatal error: camera/kona-camera.dtsi: No such file or directory
> 
> It's not like this, it's shielded or deleted.

Do you know how to find the kona-camera file?

xiangfeidexiaohuo, 2020-09-02:
> > > android/xiaomi/cmi/kernel/arch/arm64/boot/dts/qcom/kona.dtsi:5149:35: fatal error: camera/kona-camera.dtsi: No such file or directory
> > 
> > 
> > It's not like this, it's shielded or deleted.
> 
> Do you know how to find the kona-camera file?

These files are not required, remove the definition of camera.