MiCode/Xiaomi_Kernel_OpenSource #1292: When will Xiaomi release source code for Note 9, Redmi 9, Redmi 9c and Redmi 9A?

Evan-Ferrao, 2020-08-31:
It's been literally 5 months since the release of Redmi note 9 and Xiaomi doesn't seem to bother to release the source code

When will they actually release it?
And is it possible to bring this to their attention?

Vadxa, 2020-09-01:
then when and on mi not 10 lite

Evan-Ferrao, 2020-09-01:
> when

Yes, Xiaomi should release their source code also
Nowadays Xiaomi doesn't release source code as often as they used to

I really miss the old Xiaomi days when developers used to be anxious about the source code and Xiaomi was eager and supportive in releasing source code

berkiyo, 2020-09-10:
This. If any maintainers or Xiaomi developers can give an ETA, that'd be nice. I'm eager to tinker with this phone (merlin) and maybe publish a custom ROMs for this device.

PIPIPIG233666, 2020-09-10:
ETA: soon

Evan-Ferrao, 2020-09-10:
> ETA: soon

How do you know?
Do you work for Xiaomi?

Last year, Xiaomi announced that they would release source code within 3-6 months and give first priority, now they are just busy releasing phones and ignoring the community completely