MiCode/Xiaomi_Kernel_OpenSource #1293: Release Kernel source of Redmi Note 8 Pro Android 10 (MIUI 12)

SahilSonar, 2020-09-02:
When releasing new kernel sause?

ZIDAN44, 2020-09-02:
Yes.. Release New Kernel Source

sirsleon, 2020-09-02:
New kernel sources from MIUI 12 please Xiaomi, while you're at it, fix the new front stream for **newer** models of the rn8p..

aashil123, 2020-09-02:
RELEASE NEW KERNEL SOURCE NOW

D4rK7355608, 2020-09-02:
Oh yeah! Release!

manojpudasaini, 2020-09-02:
release new kernel source

Divyanshu000, 2020-09-02:
Release the Kraken!! I mean the new kernel source please.

zsalexander, 2020-09-02:
Release new Kernel Source!!!!

supperchym, 2020-09-02:
We need it.

alexmanu1, 2020-09-02:
Release the new kernel source !!

code-rgb, 2020-09-02:
Yes

takaminetakane001, 2020-09-02:
Release for Development.

porksoupdev, 2020-09-02:
Release kernial

GLokin666, 2020-09-02:
Release new kernel source!!

ZyCromerZ, 2020-09-02:
yes, we need new kernel source . . .

CopterTec, 2020-09-02:
Hello, please release the new source code. This is very important for me and the reason why I choose Xiaomi redmi.

0n3-70uch, 2020-09-02:
+1
Yes please show us the source.

gokus2264, 2020-09-02:
please release we need this thanks

FLAMEPOSEIDON, 2020-09-02:
Can't wait release pls release asap

gokus2264, 2020-09-02:
we buy xiaomi because of source code so please release this thanks a lot

Saiki672, 2020-09-02:
Release the kernel soruce of the redmi note 8 pro for development of the device.

gokus2264, 2020-09-02:
please release it now

xawbeenregmi, 2020-09-02:
Yes release the new kernal source

nabduljabbar98, 2020-09-02:
**Release New Kernel Source******__

ajaykadian99, 2020-09-02:
Release new kernel source asap.

tidieru, 2020-09-02:
Release it now,

NextWork123, 2020-09-02:
Release it now.

sw8butdeadly09, 2020-09-02:
Please release the kernel source of MIUI 12 REDMI NOTE 8 PRO

FishOnTheGround, 2020-09-02:
Release the new kernel source please!

erikcas, 2020-09-09:
did someone already mention  something about releasing new sources?

sw8butdeadly09, 2020-09-09:
> did someone already mention something about releasing new sources?

We are requesting..

Zendovo, 2020-09-10:
Yes please do!

porksoupdev, 2020-09-10:
kernial source plox

TTTT555, 2020-09-17:
+

bigrammy, 2020-10-01:
Please release the kernel source of MIUI 12 REDMI NOTE 8 PRO with the new front camera driver.

anojbaniya, 2021-02-04:
I can not use any messanger app for video call because the front camera doesn't work so please release the kernel source of miui 12

SahilSonar, 2021-02-14:
Now gib begonia-r-oss

520521mayuan, 2021-02-23:
My device is redmi note 8(ginkgo) and my front camera not work too. Please.

RouryR, 2021-10-17:
Please release the kernel source