MiCode/Xiaomi_Kernel_OpenSource #1298: Release Kernel Source of Redmi Note 9 Pro (International) /Joyeuse

AtlanPrime, 2020-09-12:
Please Provide Kernel Source For Redmi Note 9 Pro - Joyeuse,
The Kernel Source Of Redmi Note 9 S/Pro (Curtana ) And Redmi Note 9 Pro Max (Excalibur) was released as a Single Branch (https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/curtana-q-oss), apparently the devices have differences in some specs. 
Builds can be done (for joyeuse) with the existing kernel but they are observed to result in broken 33w charging (Which Is A Key Feature Of The Device As Advertised! kek), also some camera releated issues 

And if all devices (xiaomi-atoll)  have the same specs and sources , then why even bother releasing it in different names!
Literally a SCAM!
Either release proper sources or Stop releasing the same devices with different with new name.

#justiceformiatoll

etmatrix, 2021-11-22:
Have you tested the branch gauguin-r-oss?
I have a Redmi Note 9 Pro joyeuse with MIUI 12.5.4 the kernel is 4.14.180 commit e124ad7466a2 but I can't find it, so I'm analyzing which is the better kernel to compile, maybe lineageos.

AtlanPrime, 2022-01-11:
@etmatrix Hi, Sorry for the very delayed reply. I actually left the project a while ago and couldn't look into what happened afterwards. Sorry for any inconvenience caused! Good Day!

closed 2022-01-11 by AtlanPrime