MiCode/Xiaomi_Kernel_OpenSource #1299: #justiceformiatoll 😂

G-H-O-5-T, 2020-09-12:
Please Provide Kernel Source For Poco M2 Pro, Xiaomi Redmi Note 9s/pro/ProMax

The Kernel Source Of Redmi Note 9 s/pro (Curtana ) And Redmi Note 9 Pro Max (Excalibur)
Released As A Single Branch (https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/curtana-q-oss), Where The Device Have Differences In Atleast Some Specs 
And Result In Broken 33w Charging (Which Is A Key Feature Of The Device As Advertised), Also Some Camera Related Issues. 
It Should Be Same In Case Of Poco M2 Pro 

And Literally, If All Devices Have the Same Specs And Sources, Then Why You Releasing It In Different Names!
SCAM!
either release proper sources 
or 
stop releasing the same devices with different color and a new name :3

TheMalachite, 2020-09-12:
> Please Provide Kernel Source For Poco M2 Pro, Xiaomi Redmi Note 9s/pro/ProMax
>
> The Kernel Source Of Redmi Note 9 s/pro (Curtana ) And Redmi Note 9 Pro Max (Excalibur)
> Released As A Single Branch [...]

lmao what, you know curtana-q-oss is for gram, curtana, joyeuse and excalibur ?

Also don't blame Xiaomi source since miatoll custom roms litterally commonize 4 devices in one build and in one kernel defconfig which are obviously faulty for those kind of issues

yudiwdynto, 2020-09-14:
> Please Provide Kernel Source For Poco M2 Pro, Xiaomi Redmi Note 9s/pro/ProMax
>
> The Kernel Source Of Redmi Note 9 s/pro (Curtana ) And Redmi Note 9 Pro Max (Excalibur)
> Released As A Single Branch [...]

lmao