MiCode/Xiaomi_Kernel_OpenSource #1302: Mi Note 10 Lite (toco) kernel source request

sobachkamin, 2020-09-17:
Hey. Please, add source for «toco» (Mi Note 10 Lite) device. 
This wonderful device must find your support. XDA developers can create a bug free custom ROMs for the device.
Don't leave your ordinary users and fans!😃
Thanks u!

yhanz03, 2020-09-26:
maybe sir we can write an email regarding this topic here service.global@xiaomi.com. this device is useless without the sources. they can't even fix the hard touch issue and the proximity sensor issue.

marbalexbriones, 2020-10-26:
Xiaomi must listen to the requests of the users as soon as possible, many of us are waiting for this source code to be able to cook custom ROMs.

kwstpap, 2020-10-26:
xiaomi must release it past a lot of time

b0mb, 2020-11-26:
Pls release the kernel for the community ;)

moblue289, 2020-12-31:
please Xiaomi. i love your phonees and specially for custom Roms. please provide us with kernel source

EduardMcfly, 2021-02-17:
Please release the kernel for the community

sachinn360, 2021-02-18:
[https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/toco-q-oss](url)