MiCode/Xiaomi_Kernel_OpenSource #1309: Kernel source for Poco X3 (Surya/Karna)

ManoloRey, 2020-10-05:
We need this, the community has a lot to grow in terms of development! Thanks you!

UBREVO, 2020-10-30:
Please release the kernel source code already, you have stated that you would do so upon the launch of this device and failed to deliver your promise to the Poco X3 community. This device has a lot of potential, all we need now is the kernel source and we're good to go. Please make this happen, thank you.

cientista, 2020-11-12:
Another user here who choose this phone over others not just because of the specs but, because of the promise of kernel sources at launch. I decided for this one when I watch the POCO presentation and saw the images posted above.