MiCode/Xiaomi_Kernel_OpenSource #1331: Please release the kernel source code for Poco X3/NFC.

thorxone, 2020-10-28:


vasanthdeveloper, 2020-10-29:
Also, allow unlocking the bootloader for Poco X3 "karna" as it is currently not working.

UBREVO, 2020-10-30:
Please release the kernel source code already, you have stated that you would do so upon the launch of this device and failed to deliver your promise to the Poco X3 community. This device has a lot of potential, all we need now is the kernel source and we're good to go. Please make this happen, thank you.

marines, 2020-10-31:
It's not even `month one` anymore.

vasanthdeveloper, 2020-10-31:
will they even release it? 🤔

DeviantUK, 2020-10-31:
[image]
[image]

UBREVO, 2020-10-31:
> From the official POCO X3 NFC Global Launch Event:
> [image]
> [image]

Xiaomi you posted these images, why state something you weren't going to do in the first place, unless your intentions were to post on launch date and something happened which prevented you guys to upload, then you can at least give the people who are waiting a heads up? This is unacceptable, what you're doing. You're toying with those who are waiting, all we want is a heads up, that you will deliver your promise.

NotesOfReality, 2020-11-02:
**"Poco X3/NFC kernel sourcecode release"** waiting room
[image]

NotesOfReality, 2020-11-02:
(It's not like we already haven't asked for the kernel sourcecode #1309 or we aren't still asking for it #1334)

UBREVO, 2020-11-03:
> It's not like we already haven't asked for the kernel sourcecode #1309 or we aren't still asking for it #1334 btw

At least some of us are making the effort of requesting for the sources. I've looked around and you did not request or even bothered stating your reasons as to why they should release source code. Xiaomi stated they would release upon launch, they failed to deliver their promise. It's not like there's hundreds of issues being opened, there's only 3. If you look elsewhere, you'll see Mi 10 Lite kernel sources, there are more requests over there than here, so obviously our issues need to be above theirs. Just keep in mind that Xiaomi made a statement for Poco X3 and they need to deliver it.

ZjemCiKolege, 2020-11-04:
maybe in the coming months we will see sources for the Mi10T apollo :) or poco x3

UBREVO, 2020-11-04:
> maybe in the coming months we will see sources for the Mi10T apollo :) or poco x3

I hope so, going to start baking a kernel asap the moment it's released 💯

vasanthdeveloper, 2020-11-04:
Also, they still didn't release 🙄 the stock ROM for Poco X3 karna yet.

NotesOfReality, 2020-11-04:
> 
> 
> > It's not like we already haven't asked for the kernel sourcecode #1309 or we aren't still asking for it #1334 btw
> 
> At least some of us are making the effort of requesting for the sources. I've looked around and you did not request or even bothered stating your reasons as to why they should release source code. Xiaomi stated they would release upon launch, they failed to deliver their promise. It's not like there's hundreds of issues being opened, there's only 3. If you look elsewhere, you'll see Mi 10 Lite kernel sources, there are more requests over there than here, so obviously our issues need to be above theirs. Just keep in mind that Xiaomi made a statement for Poco X3 and they need to deliver it.
> 
> Edit: It's also one of the main reasons why I purchased this device, because they stated they would release kernel sources upon launch. What's not to love when they say that? Poco X3 has huge potential, it's already good as it is, but if we have the sources, just imagine how much of a beast this device can actually become.

My bad @UBREVO  , i didn't mean to disincentivize anyone, it was a way to express my disappointment towards Poco/whoever manages this repo since they should have OSSed the kernel since almost 2 months ago according to their promises, and I think a month and a half is quite enough of a "tolerance" waiting time.
I never happened to bother with making issues in Xiaomi Kernel OSS repo cause everytime I unlocked a Xiaomi device the kernel already was OSSed, hence I didn't exactly know how to move.
I honestly think too Poco X3 has great potential btw.
I'll open an issue within the weekend.

UBREVO, 2020-11-05:
> My bad @UBREVO , i didn't mean to disincentivize anyone, it was a way to express my disappointment towards Poco/whoever manages this repo since they should have OSSed the kernel since almost 2 [...]

Let's just pray that they release it this month, I'm more than willing to wait it out this month - just eager to make a custom kernel already. Honestly, the day they release the sources for this device, is the day I'll be excited 😂

mizuhashikuniumi, 2020-11-08:
Up

catfluoride, 2020-11-10:
As a new Poco X3 user, I'd love kernel source to be released.
Thanks!

cientista, 2020-11-12:
Another user here who choose this phone over others not just because of the specs but, because of the promise of kernel sources at launch. I decided for this one when I watch the POCO presentation and saw the images posted above.

vladi-it, 2020-11-13:
no kernel no poco
see you at nord

mizuhashikuniumi, 2020-11-15:
Up

akash07k, 2020-11-15:
Just release the sources Xiaomi.
It's really strange and weird that you guys didn't bothered to release the kernel sources even after more than 2 months of device release.

drumed, 2020-11-18:
Please release the kernel source code for Poco X3/NFC.

drumed, 2020-11-18:
Pls Release the Kraken

mizuhashikuniumi, 2020-11-19:
Yp

mizuhashikuniumi, 2020-11-19:
Up

mizuhashikuniumi, 2020-11-19:
Breake promise Will get karma

akza07, 2020-11-19:
This is the Karma for getting POCO X3 and believing Xiaomi's words.

drumed, 2020-11-19:
Xiaomi, keep your words

mizuhashikuniumi, 2020-11-20:
Alr released lmao

NotesOfReality, 2020-11-20:
> Alr released lmao

More like : "just" released

drumed, 2020-11-20:
Yeahhhhhhhhh, Thanks Xiaomi/Poco

thorxone, 2020-11-24:
Released

closed 2020-11-24 by thorxone