MiCode/Xiaomi_Kernel_OpenSource #1332: [REQUEST] Redmi Note 8 (Ginkgo) Android 10 Kernel Source - ginkgo-q-oss

ghostrider-reborn, 2020-10-30:
After a long wait MIUI 12 global stable has finally released for ginkgo. Please release the android 10 kernel source (**ginkgo-q-oss**) along with audio-kernel sources ASAP, complying GPL so that we can keep our kernels updated with the stock firmware. Thank you.

johnmart19, 2020-10-30:
Release sourses for Redmi note 8!

johnmart19, 2020-10-30:
And for Willow (RN8T) also!

xmxrtyx, 2020-10-30:
> After a long wait MIUI 12 global stable has finally released for ginkgo. Please release the android 10 kernel source (**ginkgo-q-oss**) along with audio-kernel sources ASAP, complying GPL so that we can keep our kernels updated with the stock firmware. Keep in mind that ginkgo-p-oss kernel sources were incomplete as audio-kernel was never uploaded for the same. Thank you.

+++

Introdructor, 2020-10-30:
Dear Xiaomi,
Can we know when it is released for this fantastic device?

Dialynx, 2020-10-30:
> After a long wait MIUI 12 global stable has finally released for ginkgo. Please release the android 10 kernel source (**ginkgo-q-oss**) along with audio-kernel sources ASAP, complying GPL so that we can keep our kernels updated with the stock firmware. Keep in mind that ginkgo-p-oss kernel sources were incomplete as audio-kernel was never uploaded for the same. Thanks in advance.

+1

aryaman895, 2020-10-31:
> After a long wait MIUI 12 global stable has finally released for ginkgo. Please release the android 10 kernel source (**ginkgo-q-oss**) along with audio-kernel sources ASAP, complying GPL so that we [...]

+1
They should release sources for ginkgo..

kutemeikito, 2020-11-05:
> After a long wait MIUI 12 global stable has finally released for ginkgo. Please release the android 10 kernel source (**ginkgo-q-oss**) along with audio-kernel sources ASAP, complying GPL so that we [...]

+1

NitroFuN, 2020-11-14:
@ivan19871002 @liuguo09 please release Q kernel source code