MiCode/Xiaomi_Kernel_OpenSource #1338: [REQUEST] Redmi Note 7 (lavender) Android 10/Q Kernel Source

z4nyx, 2020-11-13:
I have waited since January for Xiaomi to release the kernel source for lavender. Now MiUi 12 Global Stable have been released a few weeks ago. I hope you will release it ASAP. Thanks 👍🏼

akza07, 2020-11-14:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/lavender-p-oss

Isn't this lavender? Check before you Request.

z4nyx, 2020-11-14:
> https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/lavender-p-oss
> 
> Isn't this lavender? Check before you Request.

Bruh i want lavender-q-oss not lavender-p-oss ;-;

ghostrider-reborn, 2020-11-15:
> https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/lavender-p-oss
> 
> Isn't this lavender? Check before you Request.

Title of this issue says Android 10/Q Kernel Source. You have linked lavender-**p**-oss. Check before you comment.

akza07, 2020-11-17:
My bad. I just want my POCO X3 NFC to get anything. We have nothing so far.