MiCode/Xiaomi_Kernel_OpenSource #1353: 编译安卓10版本小米9内核出错

JonahBare, 2020-11-25:
严格按照wiki上小米9的编译方法编译的，clang是snapdragon-llvm-8.0.6里的，toolchain是https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9，看了好多issue，感觉可能是dtc的问题，然而具体不知道是什么问题，日志如下：

make[1]: Entering directory '/home/angie/Desktop/cepheus-q-oss/out'
  GEN     ./Makefile
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  GEN     ./Makefile
  WRAP    arch/arm64/include/generated/uapi/asm/errno.h
  WRAP    arch/arm64/include/generated/uapi/asm/ioctl.h
  WRAP    arch/arm64/include/generated/uapi/asm/ioctls.h
  WRAP    arch/arm64/include/generated/uapi/asm/ipcbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/kvm_para.h
  WRAP    arch/arm64/include/generated/uapi/asm/mman.h
  WRAP    arch/arm64/include/generated/uapi/asm/msgbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/poll.h
  WRAP    arch/arm64/include/generated/uapi/asm/resource.h
  WRAP    arch/arm64/include/generated/uapi/asm/sembuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/shmbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/socket.h
  WRAP    arch/arm64/include/generated/uapi/asm/sockios.h
  WRAP    arch/arm64/include/generated/uapi/asm/swab.h
  WRAP    arch/arm64/include/generated/uapi/asm/termbits.h
  WRAP    arch/arm64/include/generated/uapi/asm/termios.h
  WRAP    arch/arm64/include/generated/uapi/asm/types.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/basic/bin2c
  Using .. as source for kernel
  WRAP    arch/arm64/include/generated/asm/bugs.h
  WRAP    arch/arm64/include/generated/asm/clkdev.h
  WRAP    arch/arm64/include/generated/asm/delay.h
  WRAP    arch/arm64/include/generated/asm/div64.h
  WRAP    arch/arm64/include/generated/asm/dma.h
  WRAP    arch/arm64/include/generated/asm/dma-contiguous.h
  WRAP    arch/arm64/include/generated/asm/early_ioremap.h
  WRAP    arch/arm64/include/generated/asm/emergency-restart.h
  WRAP    arch/arm64/include/generated/asm/hw_irq.h
  WRAP    arch/arm64/include/generated/asm/irq_regs.h
  WRAP    arch/arm64/include/generated/asm/kdebug.h
  WRAP    arch/arm64/include/generated/asm/kmap_types.h
  WRAP    arch/arm64/include/generated/asm/local.h
  WRAP    arch/arm64/include/generated/asm/local64.h
  WRAP    arch/arm64/include/generated/asm/mcs_spinlock.h
  WRAP    arch/arm64/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/arm64/include/generated/asm/msi.h
  WRAP    arch/arm64/include/generated/asm/preempt.h
  WRAP    arch/arm64/include/generated/asm/qrwlock.h
  WRAP    arch/arm64/include/generated/asm/rwsem.h
  WRAP    arch/arm64/include/generated/asm/segment.h
  WRAP    arch/arm64/include/generated/asm/serial.h
  WRAP    arch/arm64/include/generated/asm/set_memory.h
  WRAP    arch/arm64/include/generated/asm/sizes.h
  WRAP    arch/arm64/include/generated/asm/switch_to.h
  WRAP    arch/arm64/include/generated/asm/trace_clock.h
  WRAP    arch/arm64/include/generated/asm/unaligned.h
  WRAP    arch/arm64/include/generated/asm/user.h
  WRAP    arch/arm64/include/generated/asm/vga.h
  WRAP    arch/arm64/include/generated/asm/xor.h
  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/dtc/flattree.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/genksyms/genksyms.o
  CC      scripts/mod/empty.o
  HOSTCC  scripts/mod/mk_elfconfig
  CC      scripts/mod/devicetable-offsets.s
  HOSTCC  scripts/selinux/genheaders/genheaders
[... 3931 lines cut ...]
  AR      drivers/usb/gadget/function/usb_f_rndis.o
  AR      drivers/usb/gadget/function/usb_f_mass_storage.o
  AR      drivers/usb/gadget/function/usb_f_fs.o
  AR      drivers/usb/gadget/function/usb_f_uac2.o
  AR      drivers/usb/gadget/function/usb_f_midi.o
  AR      drivers/usb/gadget/function/usb_f_hid.o
  AR      drivers/usb/gadget/function/usb_f_mtp.o
  AR      drivers/usb/gadget/function/usb_f_ptp.o
  AR      drivers/usb/gadget/function/usb_f_audio_source.o
  AR      drivers/usb/gadget/function/usb_f_accessory.o
  AR      drivers/usb/gadget/function/usb_f_diag.o
  AR      drivers/usb/gadget/function/usb_f_cdev.o
  AR      drivers/usb/gadget/function/usb_f_ccid.o
  AR      drivers/usb/gadget/udc/udc-core.o
  AR      drivers/usb/gadget/udc/built-in.o
  AR      drivers/usb/gadget/libcomposite.o
  AR      drivers/usb/gadget/function/usb_f_qdss.o
  AR      drivers/usb/gadget/function/usb_f_gsi.o
  AR      drivers/usb/gadget/function/built-in.o
  AR      drivers/usb/gadget/built-in.o
  AR      drivers/usb/built-in.o
  AR      drivers/built-in.o
make[1]: Leaving directory '/home/angie/Desktop/cepheus-q-oss/out'
Makefile:146: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2

JonahBare, 2020-11-25:
### 补上所有命令和日志
#### 目录内容
```shell
angie@AJC:~/Desktop/cepheus-q-oss$ ls
Android.bp                       build.config.goldfish.arm     COPYING        include     lib             samples     usr
AndroidKernel.mk                 build.config.goldfish.arm64   CREDITS        init        MAINTAINERS     scripts     verity_dev_keys.x509
arch                             build.config.goldfish.mips    crypto         ipc         Makefile        security    virt
block                            build.config.goldfish.mips64  Documentation  Kbuild      mm              sound
build                            build.config.goldfish.x86     drivers        Kconfig     net             toolchain
build.config.cuttlefish.aarch64  build.config.goldfish.x86_64  firmware       kernel      README          toolchains
build.config.cuttlefish.x86_64   certs                         fs             kernel.log  README-install  tools
```
#### 环境变量
```shell
$ mkdir out
$ export ARCH=arm64
$ export SUBARCH=arm64
$ export DTC_EXT=dtc
$ export CROSS_COMPILE=${PWD}/toolchain/bin/aarch64-linux-android-
```
#### 编译
```shell
$ make O=out REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_8.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu- cepheus_user_defconfig
make[1]: Entering directory '/home/angie/Desktop/cepheus-q-oss/out'
  HOSTCC  scripts/basic/fixdep
  GEN     ./Makefile
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
arch/arm64/configs/cepheus_user_defconfig:813:warning: override: reassigning to symbol CPU_FREQ_STAT
#
# configuration written to .config
#
make[1]: Leaving directory '/home/angie/Desktop/cepheus-q-oss/out'
```
```shell
$ make -j$(nproc) O=out REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_8.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu- 2>&1 | tee kernel.log
```
后面的日志就是上面的了。

JonahBare, 2020-11-25:
日志的第1652行出现出错：
Error: ../arch/arm64/boot/dts/qcom/andromeda-pinctrl.dtsi:3.1-6 syntax error
FATAL ERROR: Unable to parse input tree
scripts/Makefile.dtbo:24: recipe for target 'arch/arm64/boot/dts/qcom/andromeda-sm8150-overlay.dtbo' failed
make[3]: *** [arch/arm64/boot/dts/qcom/andromeda-sm8150-overlay.dtbo] Error 1
../scripts/Makefile.build:676: recipe for target 'arch/arm64/boot/dts/qcom' failed
make[2]: *** [arch/arm64/boot/dts/qcom] Error 2
arch/arm64/Makefile:187: recipe for target 'dtbs' failed
make[1]: *** [dtbs] Error 2
make[1]: *** Waiting for unfinished jobs....

JonahBare, 2020-11-25:
看了好几个帖子，都是安卓P的，说是要用安卓P编译出来的dtc，话说用安卓P编译出来的dtc是什么意思呢？编译安卓Q是不是要用安卓Q的dtc？第一次编译内核，孩子要晕了(╥╯^╰╥)

LuckyStar04, 2020-11-26:
我是安卓Q，我是这么解决的
sudo apt install device-tree-compiler

phone-burner, 2020-11-29:
dtc的毛病，我记得AOSP源码树里面有dtc，试试看那个
（不过也有不少人反应源码有bug

vida1j, 2020-12-01:
Same problem here on Android Q.
Did you manage to fix?
Please tell me, really need your help.

GeekerLYP, 2020-12-06:
你好。请问可以发一下那个有编译方法的wiki的网址嘛？谢谢啦

zixijian, 2020-12-21:
我编译 gauguin 内核也出现类似问题，
现在我找了两个方案：
* mk注释掉不用的dts，sm8150是骁龙855吧，这招貌似不对你的症。
* 换arch，据说Ubuntu和Debian不知道哪里有毛病。

phone-burner, 2020-12-27:
> 我编译 gauguin 内核也出现类似问题，
> 现在我找了两个方案：
> 
> * mk注释掉不用的dts，sm8150是骁龙855吧，这招貌似不对你的症。
> * 换arch，据说Ubuntu和Debian不知道哪里有毛病。

Arch与Debian/Ubuntu的差别是头文件吗？

zixijian, 2021-01-12:
> > 我编译 gauguin 内核也出现类似问题，
> > 现在我找了两个方案：
> > 
> > * mk注释掉不用的dts，sm8150是骁龙855吧，这招貌似不对你的症。
> > * 换arch，据说Ubuntu和Debian不知道哪里有毛病。
> 
> Arch与Debian/Ubuntu的差别是头文件吗？

改用tc-build项目的clang，稍微修改下报错源码，我的内核就编译出来了，据说内核4.14+就内核打了clang补丁，另wiki那里不需指定dts。

FloatingGhost, 2021-01-14:
I also had  very similar issue in gauguin, adding `CONFIG_BUILD_ARM64_DT_OVERLAY=y` to your defconfig seems to fix the issue