MiCode/Xiaomi_Kernel_OpenSource #1369: redmi note8 pro 内核编译错误

yybbll, 2020-12-11:
dear xiaomi team,
 编译 export ARCH=arm64
export SUBARCH=arm64
export CROSS_COMPILE=/home/baolin/disk2/googlesrc/aosp/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/bin/aarch64-linux-android-
make begonia_user_defconfig  O=outp   
make -j4  O=outp
编译报错：
  DTC     arch/arm64/boot/dts/mediatek/begonia-mt6785-overlay.dtb
FDVT: Drv use 5.0 folder
..
In file included from ../arch/arm64/boot/dts/mediatek/begonia-mt6785.dtsi:4:0,
                 from ../arch/arm64/boot/dts/mediatek/begonia-mt6785-overlay.dts:15:
../arch/arm64/boot/dts/mediatek/xiaomi-mt6785-common.dtsi:703:30: fatal error: k85v1_64/cust.dtsi: No such file or directory
 #include <k85v1_64/cust.dtsi>
                              ^
compilation terminated.
make[2]: *** [arch/arm64/boot/dts/mediatek/begonia-mt6785-overlay.dtb] Error 1
make[1]: *** [mediatek/begonia-mt6785-overlay.dtb] Error 2
make[1]: *** Waiting for unfinished jobs....
在源码中确实也没有找到k85v1_64/cust.dtsi 文件。请问那里配置错了呢

nbptcaot, 2021-01-05:
我也是一样的错误。。。。是不是官方传漏了文件

bzbk, 2022-07-30:
去这个网站找找 MTK的驱动 https://4pda.to/forum/index.php?showtopic=583114&st=18820 试试