MiCode/Xiaomi_Kernel_OpenSource #1394: [REQ]Android 10 kernel sources for tucana

chemusqui1, 2020-12-28:
Hello. We need the Android 10 kernel sources for Mi CC9 Pro / Mi Note 10, for having proper custom ROMs. Your p kernel source is broken on black cameras, so please update the kernel to the latest stable version and comply with GPL, thanks

iakindev, 2020-12-28:
Hey man,
Can you tell me how exactly did you compiled it?
I compiled several times but it didn't even get past the mi logo.

Either I'm compiling wrong or packing the Image-dtb wrong.

chemusqui1, 2020-12-31:
At this moment i'm stuck using prebuilt kernel binary and dtb (using the source is useless, the main issue are back cameras, dead), so this is the reason i'm asking for proper source