MiCode/Xiaomi_Kernel_OpenSource #2160: 编译 GAUGUIN 内核失败

zixijian, 2021-01-01:
环境是 红米note9pro gauguin 本机的 chroot 容器，Ubuntu18。
gcc4.8报错：  
```
9pro:~/gauguin-q-oss# make -j8 O=out ARCH=arm64 SUBARCH=arm64 DTC_EXT=dtc CROSS_COMPILE=/usr/bin/aarch64-linux-gnu-
make[1]: Entering directory '/root/gauguin-q-oss/out'
  GEN     ./Makefile
  Using .. as source for kernel
  CALL    ../scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  UPD     kernel/config_data.h
  CC      kernel/taskstats.o
  CC      kernel/tsacct.o
  CC      kernel/tracepoint.o
../kernel/taskstats.c:479:6: error: #elif with no expression
 #elif
      ^
../scripts/Makefile.build:338: recipe for target 'kernel/taskstats.o' failed
make[2]: *** [kernel/taskstats.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/root/gauguin-q-oss/Makefile:1139: recipe for target 'kernel' failed
make[1]: *** [kernel] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** wait: No child processes.  Stop.
Makefile:146: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2
```
gcc7
```
9pro:~/gauguin-q-oss# make -j8 O=out ARCH=arm64 SUBARCH=arm64 DTC_EXT=dtc CROSS_COMPILE=/usr/bin/aarch64-linux-gnu-
make[1]: Entering directory '/root/gauguin-q-oss/out'
  GEN     ./Makefile
  Using .. as source for kernel
  CALL    ../scripts/checksyscalls.sh
  CHK     include/generated/compile.h
make[3]: *** No rule to make target 'arch/arm64/boot/dts/qcom/apq8096-db820c.dtb', needed by '__build'.  Stop.
make[3]: *** Waiting for unfinished jobs....
../scripts/Makefile.build:643: recipe for target 'arch/arm64/boot/dts/qcom' failed
make[2]: *** [arch/arm64/boot/dts/qcom] Error 2
arch/arm64/Makefile:159: recipe for target 'dtbs' failed
make[1]: *** [dtbs] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/root/gauguin-q-oss/out'
Makefile:146: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2
```
注释掉dts后vm还报错，应该是ld版本高了。
就必须x86平台交叉编译才能编译小米的内核么？

zixijian, 2021-01-02:
编译去掉编译dtc参数则报错：
```
WARNING: EXPORT symbol "gsi_write_channel_scratch" [vmlinux] version generation failed, symbol will not be versioned.
/usr/bin/aarch64-linux-gnu-ld: drivers/platform/msm/gsi/gsi.o: relocation R_AARCH64_ABS32 against `__crc_gsi_write_channel_scratch' can not be used when making a shared object
drivers/platform/msm/gsi/gsi.o:/root/gauguin-q-oss/out/../drivers/platform/msm/gsi/gsi.c:4545:(.data+0x0): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(.data+0x28): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(.data+0x50): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x0): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x8): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x10): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x18): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x28): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x30): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x38): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x40): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x50): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x58): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x60): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x68): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x78): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x80): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x88): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x90): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0xa0): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0xa8): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0xb0): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0xb8): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0xc8): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0xd0): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0xd8): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0xe0): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0xf0): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0xf8): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x100): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x108): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x118): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x120): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x128): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x130): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x140): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x148): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x150): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x158): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x168): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x170): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x178): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x180): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x190): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x198): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x1a0): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x1a8): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x1b8): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x1c0): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x1c8): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x1d0): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x1e0): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x1e8): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x1f0): dangerous relocation: unsupported relocation
[... cut ...]
drivers/platform/msm/gsi/gsi.o:(__verbose+0x828): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x830): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(__verbose+0x838): dangerous relocation: unsupported relocation
/root/gauguin-q-oss/Makefile:1119: recipe for target 'vmlinux' failed
make[1]: *** [vmlinux] Error 1
make[1]: Leaving directory '/root/gauguin-q-oss/out'
Makefile:146: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2
```

zixijian, 2021-01-02:
貌似是binutils版本2.30，默认版本就是这个

zixijian, 2021-01-12:
改用TC-BUILD项目clang，稍微改一下出来了。

zixijian, 2021-01-19:
顺便贴一下issue，工程师看了可以修理一下

解决方案参考资料：

[symbol](https://github.com/Linux-On-Sdm6Series/Linux/commit/49c2b1767f601e919003a28c5ab588de7262ac0d)

[spcpy](https://github.com/kdrag0n/proton-clang/blob/master/README.md#undefined-reference-to-stpcpy)

```
MODPOST vmlinux.o
WARNING: EXPORT symbol "gsi_write_channel_scratch" [vmlinux] version generation failed, symbol will not be versioned.
ld.lld: error: relocation R_AARCH64_ABS32 cannot be used against symbol __crc_gsi_write_channel_scratch; recompile with -fPIC
>>> defined in built-in.a(drivers/platform/msm/gsi/gsi.o)
>>> referenced by gsi.c
>>>               drivers/platform/msm/gsi/gsi.o:(___kcrctab+gsi_write_channel_scratch+0x0) in archive built-in.a

ld.lld: error: undefined symbol: stpcpy
>>> referenced by tty_io.c:1137 (/root/gauguin-q-oss/out/../drivers/tty/tty_io.c:1137)
>>>               drivers/tty/tty_io.o:(alloc_tty_struct) in archive built-in.a
>>> referenced by tty_io.c:1137 (/root/gauguin-q-oss/out/../drivers/tty/tty_io.c:1137)
>>>               drivers/tty/tty_io.o:(tty_register_device_attr) in archive built-in.a
>>> referenced by tty_io.c:1137 (/root/gauguin-q-oss/out/../drivers/tty/tty_io.c:1137)
>>>               drivers/tty/tty_io.o:(show_cons_active) in archive built-in.a
>>> did you mean: strcpy
>>> defined in: lib/lib.a(string.o)
/root/gauguin-q-oss/Makefile:1116: recipe for target 'vmlinux' failed
make[1]: *** [vmlinux] Error 1
make[1]: Leaving directory '/root/gauguin-q-oss/out'
Makefile:146: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2
```

YumeMichi, 2021-01-19:
为啥不用Wiki写的流程编译呢？你用clang 12编译错误肯定得自己解决的

zixijian, 2021-01-19:
> 为啥不用Wiki写的流程编译呢？你用clang 12编译错误肯定得自己解决的

严格按照流程其实也是有问题的，我不信你按流程能编译出每一个分之。
再一个就是手机上没有那个工具链，骁龙 clang 只有x86版本的，而且只有开发者能下。

只要你别说你干嘛要用手机编译就行。
要是小米自己就做好了，哪个用户闲的跑过来自定义内核？

而且实际上我已经编译出来了，只是把 log 贴在这里方便搜索引擎收录。

closed 2021-01-19 by zixijian

CloudMeiyu, 2023-01-27:
@zixijian 兄弟，我遇到了同样的
ld.lld: error: relocation R_AARCH64_ABS32 cannot be used against symbol __crc_gsi_write_channel_scratch; recompile with -fPIC
>>> referenced by gsi.c
>>>               drivers/platform/msm/gsi/gsi.o:(___kcrctab+gsi_write_channel_scratch+0x0) in archive built-in.a
这个错误，参考资料里 “gsi”页面报错了，请问您是怎么解决这个错误的呢

SOFENGYOUNG, 2023-09-26:
@CloudMeiyu 我也遇到了ld.lld: error: relocation R_AARCH64_ABS32 cannot be used against symbol __crc_gsi_write_channel_scratch; recompile with -fPIC。 gsi那个页面不见了。兄弟你解决了吗？

zixijian, 2023-09-26:
@CloudMeiyu @SOFENGYOUNG 搜索整个github的commit总能找得到。链接已更新，差不多就是这样解决的。

SOFENGYOUNG, 2023-09-26:
@zixijian 我找到一个方案，也能编译成功，但是刷入手机后不能进入系统https://review.lineageos.org/c/LineageOS/android_kernel_asus_sm8250/+/337719