MiCode/Xiaomi_Kernel_OpenSource #2186: How to compile the kernel of Mi11 standalone

nkymey, 2021-01-20:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/How-to-compile-kernel-standalone
Nothing about Mi11/kernel5.4

nkymey, 2021-01-25:
#!/bin/bash
export OUT=${PWD}/out
export ARCH=arm64
export SUBARCH=arm64
export TARGET_BUILD_VARIANT=userdebug
#export DTC_EXT=dtc
export CROSS_COMPILE=${PWD}/toolchains/aarch64-linux-android-4.9/bin/aarch64-linux-android-
export KERNEL_DEFCONFIG=venus-qgki_defconfig
#set CONFIG_BUILD_ARM64_DT_OVERLAY=y
#set BUILD_CONFIG=build.config.gki.aarch64

O=$OUT REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_8.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu- ${PWD}/scripts/gki/generate_defconfig.sh $KERNEL_DEFCONFIG
make O=$OUT REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_8.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu- vendor/$KERNEL_DEFCONFIG
make -j$(nproc) O=$OUT REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_8.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu- 2>&1 | tee kernel.log

RTXUX, 2021-02-19:
I noticed that `venus_QGKI.config` exists in `arch/arm64/configs/vendor`, maybe calling `generate_defconfig.sh` is unnecessary?

benja95, 2021-02-21:
did you manage to make the phone boot? my Mi11 only boots without touchscreen, battery and other modules

nkymey, 2021-04-13:
Suggest to use the official reposteries for kernel build:
https://github.com/MiCode/kernel_build/tree/venus-r-oss
only boots without touchscreen, battery and other modules:
--- xiaomi_touch.ko qti_battery_charger.ko audio_xxx.ko need to modprobe or insmod by init rc scripts
refer to:
/vendor/etc/init/hw/xxx.rc
/vendor/lib/modules/

> did you manage to make the phone boot? my Mi11 only boots without touchscreen, battery and other modules

xiaoyilee, 2021-04-15:
> Suggest to use the official reposteries for kernel build:
> https://github.com/MiCode/kernel_build/tree/venus-r-oss
> only boots without touchscreen, battery and other modules:
> --- xiaomi_touch.ko [...]

Can you teach me how to compile the mi11 kernel with the official configuration?

abcz316, 2021-05-07:
> Can you teach me how to compile the mi11 kernel with the official configuration?

https://source.android.com/setup/build/building-kernels?hl=zh-cn

justcodedroid, 2023-07-05:
> PWD

您好，我按照您的配置变异疑似成功了。但是没有在out/arch/arm64/boot 下发现dtb文件，请问您遇到过这个问题吗？

dolphin-cat, 2023-09-02:
did you merge the qualcom stuff?

Guilher89, 2023-10-03:
what was the tag used in the toolchain?