MiCode/Xiaomi_Kernel_OpenSource #2191: Blackshark 2/3/3s/3pro Kernel source

wertus33333, 2021-01-25:
Will the kernel source for the **Xiaomi** Blackshark 3 be released soon?

I did some research about promises from Blackshark/Xiaomi and i'm really worried!

Sources for Blackshark 2 were promised but NEVER released! see: https://forum.xda-developers.com/t/kernel-sources.3942007/ , https://forum.xda-developers.com/t/bs2-sources-out-soon.3949267/ or just use google, there are 100ths of threads where BS promised the sources!! The Phone was released about 2 years ago! https://www.gsmarena.com/xiaomi_black_shark_2-9639.php

Furthermore, Xiaomi promised to release kernel sources within three months after launch : https://www.xda-developers.com/xiaomi-aims-to-release-kernel-source-code-for-new-devices-within-3-months-after-launch/
Blackshark is Xiaomi... 
If Blackshark can't do it for whatever reason, can Xiaomi help? it is a Xiaomi phone, Xiaomi (mostly) releases kernel sources/ keeps the promise above (not a 100% though!).

Blackshark 3s was released about 5 months ago, Blackshark 3 about a year ago: https://www.gsmarena.com/xiaomi_black_shark_3s-10350.php https://www.gsmarena.com/xiaomi_black_shark_3-10110.php

Still no sources! one full year is MORE than enough to release the sources!!

All those information lead to the conclusion that Blackshark/Xiaomi is lying to their customers and scaming them with false promises! Very dishonorable!

So Xiaomi, if you dont agree that you are scaming your customers, a rebuttal of the evidence above can be achieved by publishing the kernel source codes (BS2, 3, 3s, 3pro)!

Do the customers really have to sue/report you to get what you promised? Just release the sources, that's WAY easier and produces less complications for both sides!

If you are a user, reading this and agreeing that its not right to lie to/scam your customers, feel free to make a report here:
http://gpl-violations.org/?

koumaza, 2021-02-14:
Hi @MiCode.
Give us something to mention.
Do you pretend not to see this?

osnapitzjamie, 2021-04-01:
@mi-code

wertus33333, 2021-08-01:
@mi-code any mentions? or do you continue to keep your customers in the dark?

inuya, 2023-04-19:
How to report this to google they are the owners of the os so they can force them to release the kernal !!