MiCode/Xiaomi_Kernel_OpenSource #2202: dandelion: Missing Files

wulan17, 2021-02-08:
I tried to build dandelion (Redmi 9A) kernel last night and got this logs
https://del.dog/utoxanomib.txt
Can you give us the missing files?

TheMalachite, 2021-02-08:
> I tried to build dandelion (Redmi 9A) kernel last night and got this logs
> https://del.dog/utoxanomib.txt
> Can you give us the missing files?

Remove this line https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/dandelion-q-oss/.gitignore#L21

wulan17, 2021-02-08:
still same

wulan17, 2021-02-08:
there should be a folder named include here https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/dandelion-q-oss/drivers/input/touchscreen/mediatek/ft8006s_spi

TheMalachite, 2021-02-08:
> there should be a folder named 'include' here https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/dandelion-q-oss/drivers/input/touchscreen/mediatek/ft8006s_spi

https://github.com/TheMalachite/MiCode_kernel/commit/43493126ef405a157b4965a5caf8e77b7faef9f3

wulan17, 2021-02-09:
Ok thanks

closed 2021-02-09 by wulan17