MiCode/Xiaomi_Kernel_OpenSource #2238: cepheus-q-oss compiles successfully, but does not boot (stuck at MIUI logo)

sysedwinistrator, 2021-04-11:
Hello,
I am trying to compile and run the kernel for my Mi 9 running MIUI 12.0.4.0 QFAEUXM. Ultimately, I want to make some customizations to the kernel, but for now I am just trying to get it working with the stock config (cepheus_user_defconfig + CONFIG_BUILD_ARM64_DT_OVERLAY)
My host system is an up to date Arch Linux.

I did try to follow the [wiki's instructions](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/How-to-compile-kernel-standalone). After all, its example for msm-4.14 is the Mi 9, however for the pie branch.
The dtc was built from [aosp](https://android.googlesource.com/platform/external/dtc/), master branch (I had to change the include dtc-parser.h to dtc-parser.tab.h in dtc-lexer.l and remove acpi from libfdt in order to build it).
Both v6.0.9 (used in the wiki's example) and v8.0.6 of the Snapdragon LLVM Compiler compile without error.
But there was a catch: If only REAL_CC was specified, the build system would apparently try to use gcc and fail because of some feature (stack protection) not being supported by the compiler. If the distro clang (version 11.1.0) was specified as CC, compilation would fail at some point. The compilation would only succeed if Qualcomms clang was specified both as CC and REAL_CC.
Google's cross-compile toolchain was configured as per the wiki.
I tried flashing the kernel (both the v6.0.9 and v8.0.6 build) via AnyKernel+TWRP and bootimgtool+fastboot, both with dtbo.img and without - the system would always hang after the MIUI animation.

hendramanu, 2021-04-20:
Stack protector issue one is because either wrong compiler path or your Makefile needs some patch.

See: https://github.com/kdrag0n/proton-clang#common-problems

sysedwinistrator, 2022-03-22:
No, I ended up using a custom ROM and compiling the kernel from their source. I can only guess that the source in this repo is either outdated, or has to be built in some way that is not documented.

hjc19901119, 2023-05-16:
Hi,@2Y2s1mple .Same error.
Could you please tell me how compile and boot success?

hjc19901119, 2023-05-16:
Hi,@2Y2s1mple .Same error.
Could you please tell me how compile and boot success? 

above error info:

05-16 20:14:40.731  1186  1186 I /vendor/bin/adsprpcd: remote_handle_open: Successfully opened handle 0x64 for '":;./\createstaticpd:audiopd on domain 0
05-16 20:14:40.731  1186  1186 E /vendor/bin/adsprpcd: Error 0xffffffff: apps_dev_init failed for domain 0, errno Transport endpoint is not connected
05-16 20:14:40.731  1186  1186 E /vendor/bin/adsprpcd: Error 0xffffffff: open_dev (-1) failed for domain 0
05-16 20:14:40.732  1186  1186 E /vendor/bin/adsprpcd: vendor/qcom/proprietary/commonsys-intf/adsprpc/src/adsprpcd.c:48:adsp daemon will restart after 25ms...
05-16 20:14:40.756  1185  1185 I /vendor/bin/adsprpcd: remote_handle_open: Successfully opened handle 0x64 for '":;./\attachguestos on domain 0
05-16 20:14:40.756  1185  1185 E /vendor/bin/adsprpcd: Error 0xffffffff: apps_dev_init failed for domain 0, errno Transport endpoint is not connected
05-16 20:14:40.756  1185  1185 E /vendor/bin/adsprpcd: Error 0xffffffff: open_dev (-1) failed for domain 0
05-16 20:14:40.757  1185  1185 E /vendor/bin/adsprpcd: vendor/qcom/proprietary/commonsys-intf/adsprpc/src/adsprpcd.c:48:adsp daemon will restart after 25ms...
05-16 20:14:40.757  1186  1186 I /vendor/bin/adsprpcd: remote_handle_open: Successfully opened handle 0x64 for '":;./\createstaticpd:audiopd on domain 0

liumengdeqq, 2023-08-29:
> 

按照文档编辑mi9 和android10 系统的 Image-dbt 文件烧写到手机就会出现这种问题