MiCode/Xiaomi_Kernel_OpenSource #2240: It's about time somebody sues you

Seplx, 2021-04-24:
GPL violation is not just immoral and rude, it's also illegal. You don't include your git history, you release outdated kernel sources that don't work with new firmware (of yours), you release kernels with missing drivers, you release kernels that don't work and sometimes you don't even release the kernel source code at all! All of these violate the GPL license and all of these are a valid reason to sue you. You have been doing this for so long. Do you really think that you are not going to suffer any consequences?

bukyapavankumar, 2021-04-24:
Rightly said they need to be sued

methuselahin, 2021-05-17:
Well, Now even Joe Biden excluded u from barring list. I Guess you should be thrown back into it. Stop violating GPL