MiCode/Xiaomi_Kernel_OpenSource #2247: [Request]Force feedback(vibration) for playstations controllers

feiplane, 2021-05-14:
Please enable the force feedback(vibration) feature for playstations controllers on Poco X3 Pro(vayu)

it's actually in the kernel, just add in this line "CONFIG_SONY_FF=y" in the defconfig file this device is using would be enough already, i've already studied current kernel source, the force feedback for sony playstations controllers is not enabled, please enable it(yes i can do it myself but i don't want to root my x3 pro just yet), thx

adi8900, 2021-05-15:
lmao

feiplane, 2021-05-15:
> lmao

What's ur problem?

Kampalus, 2021-05-15:
> > lmao
> 
> What's ur problem?

Probably just laughing at multiple things you are overlooking:

1. No OEM cares about user feedback and/or requests
2. No one at Xiaomi ever looks at this account
3. Xiaomi is too busy releasing a new phone every two weeks thus not updating the kernel

feiplane, 2021-05-16:
> > > lmao
> > 
> > 
> > What's ur problem?
> 
> Probably just laughing at multiple things you are overlooking:
> 
> 1. No OEM cares about user feedback and/or requests
> 2. No one at Xiaomi ever looks at this account
> 3. Xiaomi is too busy releasing a new phone every two weeks thus not updating the kernel

well, at least i've tried, neglected or not, can't control over it. And in fact, i've already provided solution altogether, just few clicks away enough to do the trick(unlike others just asking). There's no problem to diy though, but the admin of x3 pro official tele group said that warranty will be affected once rooted, what else to do?