MiCode/Xiaomi_Kernel_OpenSource #2250: [REQUEST] Kernel source for Redmi K30S Ultra (apollo) Android 11

hendramanu, 2021-05-24:
Hi! Please release the Redmi K30S Ultra kernel source for Android 11 (R). There's now only Android 10 source.

Thanks!

Aquarius223, 2021-05-29:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/cmi-r-oss

hendramanu, 2021-05-29:
Woops, I mean Mi 10T

Aquarius223, 2021-05-29:
Kernel: https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/cmi-r-oss
audio-kernel: https://github.com/MiCode/vendor_qcom_opensource_audio-kernel/tree/cmi-r-oss
data-kernel: https://github.com/MiCode/vendor_qcom_opensource_data-kernel/tree/cmi-r-oss
wlan: https://github.com/MiCode/vendor_qcom_opensource_wlan/tree/cmi-r-oss

defconfig: apollo_user_defconfig

hendramanu, 2021-05-29:
Woaah, thank you so much! Don't know about that. Closing it

closed 2021-05-29 by hendramanu