MiCode/Xiaomi_Kernel_OpenSource #2263: Release kernel source for begonia-r-oss

SahilSonar, 2021-06-19:
We want kernel source for begonia-r-oss

sirsleon, 2021-06-19:
Yeah, do it and comply with GPL V3 pls.

manojpudasaini, 2021-06-19:
We want **Kernel Source** for begonia-r-oss

Dipanshu6, 2021-06-19:
yes please, fast af

iitachiuchiha, 2021-06-19:
Release Kernel Source for begonia-r-oss

stellabyte2, 2021-06-19:
We want kernel source for begonia-r-oss

Jithin564, 2021-06-19:
Please release kernal source for begonia-r-oss as soon as possible.

neebe2, 2021-06-19:
Need sauce :v

santoshhh000, 2021-06-19:
New kernel source when ...

abhixv, 2021-06-19:
We want begonia-r-oss kernel source

Droid-USSR, 2021-06-19:
We want kernel source for begonia-r-oss

neebe2, 2021-06-19:
all pro here , except me :D

DrtSinX98, 2021-06-19:
Give kernel source for begonia-r-oss ;-;

naveeneppalapally, 2021-06-19:
We want kernel source for begonia-r-oss

HyperNotAryanX97, 2021-06-19:
We want kernel source for begonia-r-oss

ImperfectNiBos, 2021-06-19:
We need kernel source for begonia-r-oss

Flashokilleralt, 2021-06-19:
We want Kernel Source for begonia-r-oss

Flashokilleralt, 2021-06-19:
#We want Kernel Source for begonia-r-oss give plox don't violate GPL

FLAMEPOSEIDON, 2021-06-19:
give begonia r-oss kernel Xiaomi

Potexmachine, 2021-06-19:
Please give begonia-r-oss kernel sources we deserve and need them 🥺🥺🥺

Hanan3000, 2021-06-19:
We want kernel source for begonia-r-oss

Talhac21, 2021-06-19:
We want kernel source for begonia-r-oss please

stellabyte2, 2021-06-19:
Madarchod kernel source de...

ios7jbpro, 2021-06-19:
yea we do want it

TheFormidable, 2021-06-19:
We need kernel source for begonia-r-oss (don't forget the new camera sensors)

Kardch, 2021-06-19:
We want kernel source for begonia-r-oss please

Myehia201, 2021-06-19:
We want kernel source for begonia-r-oss

XDinisev, 2021-06-19:
We want kennel source for begonia-r-oss

themirvel, 2021-06-19:
We want kernel source for begonia-r-oss  🥺

XDinisev, 2021-06-23:
We want kernel source for begonia-r-oss

Dv1101, 2021-06-26:
Give Begonia-r-oss kernel source

HyperNotShroudX97, 2021-06-29:
Meowdharchomd Xioameme Kernel Sauces Dedo Bhosdiwalo

KangRuwsuh, 2021-06-30:
Give us begonia-r-oss source please :")

HyperNotAryanX97, 2021-07-01:
Madharboard Kernel Source dede

HyperNotAryanX97, 2021-07-01:
Fuck You Xiaomeme Give Source.
[image]

XDinisev, 2021-07-15:
We need kernel source for begonia-r-oss

itsmeyukiedcqii, 2021-07-20:
We want Kernel Source for begonia-r-oss

leyash143, 2021-07-20:
We want the kernel source for begonia-r-oss, we need it badly! 😭🥺

compassnet, 2021-08-18:
Please release the kernel source code for begonia-r-oss. Thank you!

grimanos1, 2021-08-18:
Please release the kernel source code for begonia-r-oss. Thank you!

7Soldier, 2021-08-21:
RELEASE F*CKIN BEGONIA-R-OSS. XIAOMI PEACE OF SHIT. WHERE THE OUR SOURCES?!