MiCode/Xiaomi_Kernel_OpenSource #2270: Missing focaltech touchscreen driver on camellia-r-oss

TheMalachite, 2021-07-11:
Hello, I noticied on stock MIUI defconfig that focaltech directory driver is **FT8722**, however that driver doesn't exist on **camellia-r-oss** which mean that touchscreen not gonna work on unit who use a focaltech touchscreen. Could you please add that driver ?

willzyx-hub, 2021-07-11:
That's right custom rom & custom kernel will be impossible without it.

duelistsage, 2021-07-11:
We really need this.

willzyx-hub, 2021-07-12:
Hey Xiaomi Do something !

itzmonvw124, 2021-07-12:
We want it so much Xiaomi do something

hibreck, 2021-07-12:
hello xiaomi, we really need this driver, we will be most grateful to you if you publish it :)

ModdingExpert1988, 2021-07-13:
Yes we really need

ModdingExpert1988, 2021-07-13:
We meee pleaseeeeeee

ModdingExpert1988, 2021-07-13:
We need please make

ModdingExpert1988, 2021-07-13:
Make

ModdingExpert1988, 2021-07-13:
Make

willzyx-hub, 2021-07-13:
Comeon Xiaomi Don't make us beg for it.

ModdingExpert1988, 2021-07-13:
Pleeeeaaeeee xiaoomiiiii

ModdingExpert1988, 2021-07-13:
Please release focaltech driver

ModdingExpert1988, 2021-07-13:
Release

duelistsage, 2021-07-13:
@mi-code

duelistsage, 2021-07-13:
@mi-code

duelistsage, 2021-07-13:
@mi-code

ModdingExpert1988, 2021-07-13:
@mi-code

duelistsage, 2021-07-13:
@liuguo09

duelistsage, 2021-07-13:
@thune-xiaobai

rex2630, 2021-07-13:
You are really retarded spammer, my email is hurting me now

itzmonvw124, 2021-07-13:
> You are really retarded spammer, my email is hurting me now
> 
> 
> > @thune-xiaobai <https://github.com/thune-xiaobai>
> >
> >
> 
> 
> With best regards:
> David
> From
> Portal4Gamers
> -----------------------------------
> http://www.mobilecowboys.nl/images/upload/1218185605Stop%20Vodafone.jpg

I'm sorry about that we need the driver so much we can't build anything if haven't it

rex2630, 2021-07-13:
I get that, but spamming wont change it, writing it one time will have same
effect

duelistsage, 2021-07-13:
#2273

willzyx-hub, 2021-07-13:
> I get that, but spamming wont change it, writing it one time will have same effect

Spam will being XiaoMeme Attention.

willzyx-hub, 2021-07-13:
> I get that, but spamming wont change it, writing it one time will have same effect

Spam will bring XiaoMi Attention. So why you don't you press unsubscribe button ?

ModdingExpert1988, 2021-07-14:
> I get that, but spamming wont change it, writing it one time will have same effect

Shut up and release the driver or will buy a samsung next time

ModdingExpert1988, 2021-07-14:
> I get that, but spamming wont change it, writing it one time will have same effect

You fucking dumb

willzyx-hub, 2021-07-14:
> > I get that, but spamming wont change it, writing it one time will have same effect
> 
> You fucking dumb

Exactly, maybe he are XiaoMeme Dev spam him to release missing driver.

willzyx-hub, 2021-07-19:
@mi-code hey XiaoMi Give US THE FKING DRIVER !!! SIMPLY RELEASE IT DON'T MAKE US BEG FOR IT OR I WILL BUY REALME NEXT TIME !!!

willzyx-hub, 2021-07-19:
您好小米我在camellia-r-oss中看到一个问题，我找不到Redmi Note 10 5G / Poco M3 Pro设备的Touch驱动程序我要求小米尽快发布驱动程序。 因为我们真的需要它，特别是驱动程序对于像 TWRP 这样的自定义恢复非常重要。 没有它，我们就无法制作自定义 rom。 谢谢你。

(Translated by Google)

willzyx-hub, 2021-07-19:
> 您好小米我在camellia-r-oss中看到一个问题，我找不到Redmi Note 10 5G / Poco M3 Pro设备的Touch驱动程序我要求小米尽快发布驱动程序。 因为我们真的需要它，特别是驱动程序对于像 TWRP 这样的自定义恢复非常重要。 没有它，我们就无法制作自定义 rom。 谢谢你。
> 
> (Translated by Google)

@mi-code

Adou2056, 2021-07-30:
I'm wondering **when the driver can be released.**
I bought this Redmi Note 10,try to compile native ROM for use...
But now I‘ve encountered the same problem.
Since this kernel project is open-source, can it be perfected?
@mi-code

Adzetko, 2021-08-17:
Hey @mi-code we are really looking for that driver, it would help us enthusiasts coders to get things done correctly. This device is MediaTek so it's already hard enough to dev custom roms and everything on it, if you guys could give us some more drivers we would all really appreciate.

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
[camellia-r-oss](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/camellia-r-oss) @mi-code

[13 more replies]