MiCode/Xiaomi_Kernel_OpenSource #2284: Missing focaltech touchscreen driver on camellia-r-oss

ModdingExpert1988, 2021-07-14:
Hello, I noticied on stock MIUI defconfig that focaltech directory driver is FT8722, however that driver doesn't exist on camellia-r-oss which mean that touchscreen not gonna work on unit who use a focaltech touchscreen. Could you please add that driver ? If you don't add the driver I will buy samsung next time

vishansh, 2021-07-14:
Bro is this driver needed for Redmi Note 10 5G or Poco M3 Pro 5G coz they haven't updated the source for Poco M3 Pro 5G

ModdingExpert1988, 2021-07-14:
> Bro is this driver needed for Redmi Note 10 5G or Poco M3 Pro 5G coz they haven't updated the source for Poco M3 Pro 5G

We need to build custom roms and twrp

iamverycute, 2021-11-17:
期待camellia触摸屏驱动