MiCode/Xiaomi_Kernel_OpenSource #2291: 编译k30 ultra cezanne时报错

flben233, 2021-07-31:
In file included from ../mm/memcontrol.c:69:0:
../include/net/sock.h:1696:1: error: 'no_sanitize' attribute directive ignored [-Werror=attributes]
 {
 ^
In file included from ../include/net/ip.h:31:0,
                 from ../mm/memcontrol.c:70:
../include/net/route.h:235:1: error: 'no_sanitize' attribute directive ignored [-Werror=attributes]
 {
 ^
  CC      drivers/base/cacheinfo.o
  CC      crypto/asymmetric_keys/public_key.o
  CC      block/cfq-iosched.o
  AR      fs/crypto/fscrypto.o
  CC      drivers/base/power/sysfs.o
  AR      fs/crypto/built-in.o
  CC      fs/devpts/inode.o
  ASN.1   crypto/asymmetric_keys/x509-asn1.c
  ASN.1   crypto/asymmetric_keys/x509_akid-asn1.c
  CC      crypto/asymmetric_keys/x509_public_key.o
  CC      drivers/base/power/generic_ops.o
  AR      fs/devpts/devpts.o
  AR      fs/devpts/built-in.o
  AR      fs/exofs/built-in.o
  CC      fs/exportfs/expfs.o
  ASN.1   crypto/asymmetric_keys/pkcs7-asn1.c
  CC      crypto/asymmetric_keys/pkcs7_trust.o
  CC      block/mq-deadline.o
cc1: all warnings being treated as errors
../scripts/Makefile.build:356: recipe for target 'mm/memcontrol.o' failed
make[2]: *** [mm/memcontrol.o] Error 1
/home/Xiaomi/Makefile:1163: recipe for target 'mm' failed
make[1]: *** [mm] Error 2
make[1]: *** Waiting for unfinished jobs....
…………
In file included from ../block/blk-crypto.c:14:0:
../block/blk-crypto.c: In function 'blk_crypto_flock':
../include/linux/keyslot-manager.h:101:13: error: inlining failed in call to always_inline 'ksm_flock': function body not available
 inline void ksm_flock(struct keyslot_manager *ksm, unsigned int flags);
             ^
../block/blk-crypto.c:310:2: error: called from here
  ksm_flock(ksm, flags);
  ^
../scripts/Makefile.build:356: recipe for target 'block/blk-crypto.o' failed
make[2]: *** [block/blk-crypto.o] Error 1
make[2]: *** Waiting for unfinished jobs....
  CC      fs/f2fs/file.o
/home/Xiaomi/Makefile:1163: recipe for target 'block' failed
make[1]: *** [block] Error 2
  AR      drivers/input/misc/aw8697_haptic/built-in.o
../drivers/gpu/drm/mediatek/mtk_disp_color.c: In function 'mtk_color_user_cmd':
../drivers/gpu/drm/mediatek/mtk_disp_color.c:3127:4: error: format '%x' expects a matching 'unsigned int' argument [-Werror=format=]
    DDPINFO("dual pipe pa:0x%x(va:0x%lx) = 0x%x (0x%x) comp->reg_pa:(0x%x)\n",
    ^
  CC      drivers/input/misc/uinput.o
  CC      drivers/leds/led-triggers.o
  CC      drivers/input/touchscreen/GT9896/goodix_ts_i2c.o
cc1: all warnings being treated as errors
../scripts/Makefile.build:356: recipe for target 'drivers/gpu/drm/mediatek/mtk_disp_color.o' failed
make[5]: *** [drivers/gpu/drm/mediatek/mtk_disp_color.o] Error 1
../scripts/Makefile.build:671: recipe for target 'drivers/gpu/drm/mediatek' failed
make[4]: *** [drivers/gpu/drm/mediatek] Error 2
../scripts/Makefile.build:671: recipe for target 'drivers/gpu/drm' failed
make[3]: *** [drivers/gpu/drm] Error 2
../scripts/Makefile.build:671: recipe for target 'drivers/gpu' failed
make[2]: *** [drivers/gpu] Error 2
make[2]: *** Waiting for unfinished jobs....
……
  AR      drivers/input/touchscreen/GT9896/goodix_core.o
../drivers/input/touchscreen/GT9896/goodix_ts_test.c: In function 'gtx8_analyse_diffdata':
../drivers/input/touchscreen/GT9896/goodix_ts_test.c:2209:6: warning: unused variable 'find_bad_node' [-Wunused-variable]
  u32 find_bad_node = 0;
      ^
  AR      drivers/input/touchscreen/GT9896/built-in.o
  AR      drivers/input/touchscreen/built-in.o
  AR      drivers/input/built-in.o
/home/Xiaomi/Makefile:1163: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
  CC [M]  kernel/kheaders.o
make[1]: Leaving directory '/home/Xiaomi/out'
Makefile:146: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2

日志地址：https://pastebin.com/TNUigpix
 日志网盘地址：https://drive.google.com/file/d/121X4KUYNSubZX2r8lKCz7M1JdYB5Gtpv/view?usp=sharing

flben233, 2021-07-31:
@ivan19871002 @mi-code

closed 2021-08-01 by flben233

2512500960, 2021-08-20:
same error, any fix yet ?
`keyslot-manager.h:101:13: error: inlining failed in call to always_inline 'ksm_flock': function body not availabl
e`

papadave66, 2021-11-24:
same error as mine

2512500960, 2021-11-24:
successfully compiled by doing so:
1. add clang-r383902 and aarch64-linux-android-4.9 toolchain to your path
2. ARCH=arm64 make CC=clang HOSTCC=gcc AR=llvm-ar NM=llvm-nm OBJCOPY=llvm-objcopy OBJDUMP=llvm-objdump STRIP=llvm-strip O=out CLANG_TRIPLE=aarch64-linux-gnu- CROSS_COMPILE=aarch64-linux-android- LD=ld.lld cezanne_user_defconfig
3. ARCH=arm64 make CC=clang HOSTCC=gcc AR=llvm-ar NM=llvm-nm OBJCOPY=llvm-objcopy OBJDUMP=llvm-objdump STRIP=llvm-strip O=out CLANG_TRIPLE=aarch64-linux-gnu- CROSS_COMPILE=aarch64-linux-android- LD=ld.lld -j10