MiCode/Xiaomi_Kernel_OpenSource #2292: [apollo-r-oss][V12.5.3.0.RJDEUXM] New source.

yEsPaP, 2021-08-05:
Hello please release source for apollo V12.5.3.0.RJDEUXM firmware kernel

i tried using source cmi-r-oss, but that source has missing things

compared to stock extracted .config from boot.img it's missing things like 
```
CONFIG_MI_RECLAIM=y
CONFIG_MIGT_KGSL=y
CONFIG_MIGT=y
CONFIG_MILLET=y
CONFIG_MIHW=y
```

So please update the source for apollo android R.
Thank you.

Aquarius223, 2021-08-05:
These contents belong to Xiaomi private library and will not be opensource

Aquarius223, 2021-08-05:
You can get part of the code in cas-q-oss
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/cas-q-oss/drivers/staging/mi-reclaim

https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/cas-q-oss/kernel

yEsPaP, 2021-08-05:
> These contents belong to Xiaomi private library and will not be opensource

interesting. isn't that against the GPLv2 license the kernel has

Aquarius223, 2021-08-09:
> > These contents belong to Xiaomi private library and will not be opensource
> 
> interesting. isn't that against the GPLv2 license the kernel has, why in other device kernel its there?

It may not be in the kernel (msm-4.19) elsewhere
My device: Xiaomi Mi 10 umi
Through dump, some drivers are found in /home/work/umi-r-stable-build/miui/kernel. Does not belong to kernel/msm-4.19