MiCode/Xiaomi_Kernel_OpenSource #2300: k30 ultra cezanne-r-oss 用什么编译工具链好？

2512500960, 2021-08-19:
rt，用clang还是aarch64-linux-gnu-6.3.1? 看到一堆build.config

2512500960, 2021-08-25:
build.config.mtk.aarch64:clang不会因为一些-Werror报错。

closed 2021-08-25 by 2512500960

Angel0726, 2022-02-06:
你是按照wiki做的吗？怎么编译的

2512500960, 2022-02-06:
https://gitee.com/erlkonig/xiaomi_kernel_cezanne_r_oss/blob/cezanne-r-oss/kernel_compile_step.txt参考一下？

Angel0726, 2022-02-06:
> https://gitee.com/erlkonig/xiaomi_kernel_cezanne_r_oss/blob/cezanne-r-oss/kernel_compile_step.txt参考一下？

谢谢

Angel0726, 2022-02-06:
> https://gitee.com/erlkonig/xiaomi_kernel_cezanne_r_oss/blob/cezanne-r-oss/kernel_compile_step.txt参考一下？

谢谢，这种方法编译成功了吗？

2512500960, 2022-02-06:
> > [https://gitee.com/erlkonig/xiaomi_kernel_cezanne_r_oss/blob/cezanne-r-oss/kernel_compile_step.txt参考一下？](https://gitee.com/erlkonig/xiaomi_kernel_cezanne_r_oss/blob/cezanne-r-oss/kernel_compile_step.txt%E5%8F%82%E8%80%83%E4%B8%80%E4%B8%8B%EF%BC%9F)
> 
> 谢谢，这种方法编译成功了吗？

ok的，

Angel0726, 2022-02-07:
> > > [https://gitee.com/erlkonig/xiaomi_kernel_cezanne_r_oss/blob/cezanne-r-oss/kernel_compile_step.txt参考一下？](https://gitee.com/erlkonig/xiaomi_kernel_cezanne_r_oss/blob/cezanne-r-oss/kernel_compile_step.txt%E5%8F%82%E8%80%83%E4%B8%80%E4%B8%8B%EF%BC%9F)
> > 
> > 
> > 谢谢，这种方法编译成功了吗？
> 
> ok的，

可以用miui12.5.7吗？内核版本号是4.14.186-g498fa9c577c2吗

2512500960, 2022-02-07:
> 可以用miui12.5.7吗？内核版本号是4.14.186-g498fa9c577c2吗

不太理解你意思，我是用它发布的内核源码编译出来我要的内核了，编译工具链用的google给的clang-r383902

Angel0726, 2022-02-07:
> > 可以用miui12.5.7吗？内核版本号是4.14.186-g498fa9c577c2吗
> 
> 不太理解你意思，我是用它发布的内核源码编译出来我要的内核了，编译工具链用的google给的clang-r383902

我用的miui12.5.7。怕编译出来的内核不能开机。clang、gcc是这两个链接吗？
git clone https://android.googlesource.com/platform/prebuilts/clang/host/linux-x86
git clone https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9

2512500960, 2022-02-07:
是的

Angel0726, 2022-02-08:
> 可以用miui12.5.7吗？内核版本号是4.14.186-g498fa9c577c2吗

我参考了这个教程。但是发现几个问题就是
```
/mnt/d/kernel/toolclang/clang-r433404/bin/clang: Command not found
/bin/sh: 1: ld.lld: not found
drivers/misc/mediatek/nfc/Kconfig:22:warning: ignoring type redefinition of 'MTK_NFC_CLKBUF_ENABLE' from 'boolean' to 'tristate'
```

Angel0726, 2022-02-08:
> 是的

十分感谢。我写了一份编译教程https://github.com/Angel0726/andriod_kernel/blob/main/k30_ultra.md。其中有个问题想咨询一下。就是part1、part2、Additional part:这三个部分都有什么区别呀？
[image]

2512500960, 2022-02-08:
> 十分感谢。我写了一份编译教程[https://github.com/Angel0726/andriod_kernel/blob/main/k30_ultra.md。其中有个问题想咨询一下。就是part1、part2、Additional](https://github.com/Angel0726/andriod_kernel/blob/main/k30_ultra.md%E3%80%82%E5%8 [...]

第一部分结束后，得到dtb文件，内核文件，手机自带的wifi,bt,gps等的驱动模块，够用了。第二部分，是编译另外两个内核模块，源码在小米的mtk module对应的仓库，目测没啥用，可以忽略。第三部分，在第一部分中得到的那个wifi驱动模块文件大小特别大，看着不舒服的话，可以手动strip一下，可以忽略。

Angel0726, 2022-02-08:
> 第一部分结束后，得到dtb文件，内核文件，手机自带的wifi,bt,gps等的驱动模块，够用了。第二部分，是编译另外两个内核模块，源码在小米的mtk module对应的仓库，目测没啥用，可以忽略。第三部分，在第一部分中得到的那个wifi驱动模块文件大小特别大，看着不舒服的话，可以手动strip一下，可以忽略。

好的十分感谢

Angel0726, 2022-02-08:
> 第一部分结束后，得到dtb文件，内核文件，手机自带的wifi,bt,gps等的驱动模块，够用了。第二部分，是编译另外两个内核模块，源码在小米的mtk module对应的仓库，目测没啥用，可以忽略。第三部分，在第一部分中得到的那个wifi驱动模块文件大小特别大，看着不舒服的话，可以手动strip一下，可以忽略。

你编译config的时候会有这个问题吗？
[image]

2512500960, 2022-02-08:
> 你编译config的时候会有这个问题吗？ [image]

好像是有，warning忽略得了

Angel0726, 2022-02-09:
> 好像是有，warning忽略得了
这个可以删掉吗？但是我看好多变量只是warning，不清楚为什么这个是error
[image]
[image]

2512500960, 2022-02-09:
> 是的 [...]

因为这个c文件被编译时，使用的编译器参数既有-wunused-but-set-variable参数（没用但是设置了变量），同时还有-Werror（警告作为错误处理），这两张图报错的时候说了的。
[image]
[image]。我印象中编译这个文件的时候没报错，你检查下clang的版本是不是一模一样的。

Angel0726, 2022-02-09:
> 因为这个c文件被编译时，使用的编译器参数既有-wunused-but-set-variable参数（没用但是设置了变量），同时还有-Werror（警告作为错误处理），这两张图报错的时候说了的。 [image] [image]。我印象中编译这个文件的时候没报错，你检查下clang的版本是不是一模一样的。

好的十分感谢

Angel0726, 2022-02-09:
> 我印象中编译这个文件的时候没报错，你检查下clang的版本是不是一模一样的。

我将clang版本修改到了r383902。但是新多了一个错误
[image]

2512500960, 2022-02-09:
> > 我印象中编译这个文件的时候没报错，你检查下clang的版本是不是一模一样的。
> 
> 我将clang版本修改到了r383902。但是新多了一个错误 [image]

emmm，你的所有操作过程发一下？从设置环境变量，到make命令，然后你参考下这个回答，https://stackoverflow.com/questions/66799641/usr-bin-as-unrecognized-option-el

Angel0726, 2022-02-09:
> > > 我印象中编译这个文件的时候没报错，你检查下clang的版本是不是一模一样的。
> > 
> > 
> > 我将clang版本修改到了r383902。但是新多了一个错误 [image]
> 
> emmm，你的所有操作过程发一下？从设置环境变量，到make命令，然后你参考下这个回答，https://stackoverflow.com/questions/66799641/usr-bin-as-unrecognized-option-el
我是用的是wsl2 ubuntu20.04
# 安装环境
```bash
# zlib1g-dev:i386、python-networkx没有安装包所以没有安装
sudo apt-get install git ccache automake flex lzop bison \
gperf build-essential zip curl zlib1g-dev  \
g++-multilib  libxml2-utils bzip2 libbz2-dev \
libbz2-1.0 libghc-bzlib-dev squashfs-tools pngcrush \
schedtool dpkg-dev liblz4-tool make optipng maven libssl-dev \
pwgen libswitch-perl policycoreutils minicom libxml-sax-base-perl \
libxml-simple-perl bc libc6-dev-i386 lib32ncurses5-dev \
x11proto-core-dev libx11-dev lib32z-dev libgl1-mesa-dev xsltproc unzip
```
# 代码
源码&编译工具gcc&编译工具clang  在同一位置.当前目录中包括以下3个文件夹
- toolgcc：需要回退到95280
- toolclang：需要回退到b800f
- Xiaomi_Kernel_OpenSource：小米源代码
```bash
# 源代码
git clone https://github.com/MiCode/Xiaomi_Kernel_OpenSource.git -b cezanne-r-oss
# 编译工具gcc   版本号 95280bf
git clone https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9 toolgcc
cd toolgcc && git reset --hard 95280   # 最新版本没有gcc。所以要回退到95280
cd .. #回退完之后，返回上一层目录
# clang  如果git clone中途中断可以添加 --recursive，添加代理http.proxy="http://127.0.0.1:1082"下载
git clone  --recursive -c http.proxy="http://127.0.0.1:1082" https://android.googlesource.com/platform/prebuilts/clang/host/linux-x86  toolclang
cd toolclang && git reset --hard b800f   # 最新版本没有r383902。所以要回退到b800f
```
# 设置环境
添加环境变量。因为使用了${PWD}命令，所以需要在原始目录添加环境变量。
```bash
export ARCH=arm64
export SUBARCH=arm64
# toolgcc中bin文件夹的位置 
export CROSS_COMPILE=${PWD}/toolgcc/bin/aarch64-linux-android-
# 添加环境变量    clang版本号b800f
export PATH=${PWD}/toolgcc/bin:${PWD}/toolclang/clang-r383902/bin:$PATH
```
# 编译
```bash
# 进入代码文件夹
cd Xiaomi_Kernel_OpenSource
make CC=clang HOSTCC=gcc AR=llvm-ar NM=llvm-nm OBJCOPY=llvm-objcopy OBJDUMP=llvm-objdump STRIP=llvm-strip O=out CLANG_TRIPLE=aarch64-linux-gnu- CROSS_COMPILE=aarch64-linux-android- LD=ld.lld cezanne_user_defconfig
make CC=clang HOSTCC=gcc AR=llvm-ar NM=llvm-nm OBJCOPY=llvm-objcopy OBJDUMP=llvm-objdump STRIP=llvm-strip O=out CLANG_TRIPLE=aarch64-linux-gnu- CROSS_COMPILE=aarch64-linux-android- LD=ld.lld -j4
```

2512500960, 2022-02-09:
吧ARCH=arm64放到make那条命令里，那个export可能不管用

Angel0726, 2022-02-09:
> 吧ARCH=arm64放到make那条命令里，那个export可能不管用

放了不管用。此外还把所有目录做了chmod 777
[image]

2512500960, 2022-02-09:
没碰上过这个，网上搜这个错误，说是调错汇编器（as）了，gnu的as支持-EL选项。那就玄学操作一拨儿呗：
1. ARCH=arm64 make clean，然后再make的时候ARCH=arm64家前面，额外再加一个LLVM_IAS=0，https://www.kernel.org/doc/html/latest/kbuild/llvm.html
2. 全部删掉，重新来

Angel0726, 2022-02-09:
> 没碰上过这个，网上搜这个错误，说是调错汇编器（as）了，gnu的as支持-EL选项。那就玄学操作一拨儿呗：
> 
> 1. ARCH=arm64 make clean，然后再make的时候ARCH=arm64家前面，额外再加一个LLVM_IAS=0，https://www.kernel.org/doc/html/latest/kbuild/llvm.html
> 2. 全部删掉，重新来

好的，我试一下

Angel0726, 2022-02-10:
> 没碰上过这个，网上搜这个错误，说是调错汇编器（as）了，gnu的as支持-EL选项。那就玄学操作一拨儿呗：
> 
> 1. ARCH=arm64 make clean，然后再make的时候ARCH=arm64家前面，额外再加一个LLVM_IAS=0，https://www.kernel.org/doc/html/latest/kbuild/llvm.html
> 2. 全部删掉，重新来
还是不行。找不到error，最后还是 `[Makefile:146: sub-make] Error 2`

2512500960, 2022-02-10:
> > 没碰上过这个，网上搜这个错误，说是调错汇编器（as）了，gnu的as支持-EL选项。那就玄学操作一拨儿呗：
> > 
> > 1. ARCH=arm64 make clean，然后再make的时候ARCH=arm64家前面，额外再加一个LLVM_IAS=0，https://www.kernel.org/doc/html/latest/kbuild/llvm.html
> > 2. 全部删掉，重新来
> >    还是不行。找不到error，最后还是 `[Makefile:146: sub-make] Error 2`

找不到错误，就在错误发生之后，再-j1单线程编译，同时加上V=s查看详细输出。

Angel0726, 2022-02-10:
> > > 没碰上过这个，网上搜这个错误，说是调错汇编器（as）了，gnu的as支持-EL选项。那就玄学操作一拨儿呗：
> > > 
> > > 1. ARCH=arm64 make clean，然后再make的时候ARCH=arm64家前面，额外再加一个LLVM_IAS=0，https://www.kernel.org/doc/html/latest/kbuild/llvm.html
> > > 2. 全部删掉，重新来
> > >    还是不行。找不到error，最后还是 `[Makefile:146: sub-make] Error 2`
> 
> 找不到错误，就在错误发生之后，再-j1单线程编译，同时加上V=s查看详细输
好多warning。我试试`ARCH=arm64 make CC=clang HOSTCC=clang AR=llvm-ar NM=llvm-nm OBJCOPY=llvm-objcopy OBJDUMP=llvm-objdump STRIP=llvm-strip O=out CLANG_TRIPLE=aarch64-linux-gnu- CROSS_COMPILE=aarch64-linux-android- LD=$source_dir/toolclang/clang-bootstrap/bin/ld.lld cezanne_user_defconfig`

Angel0726, 2022-02-12:
> > > 没碰上过这个，网上搜这个错误，说是调错汇编器（as）了，gnu的as支持-EL选项。那就玄学操作一拨儿呗：
> > > 
> > > 1. ARCH=arm64 make clean，然后再make的时候ARCH=arm64家前面，额外再加一个LLVM_IAS=0，https://www.kernel.org/doc/html/latest/kbuild/llvm.html
> > > 2. 全部删掉，重新来
> > >    还是不行。找不到error，最后还是 `[Makefile:146: sub-make] Error 2`
> 
> 找不到错误，就在错误发生之后，再-j1单线程编译，同时加上V=s查看详细输出。

还是不行。应该是编译器的问题。你能帮我看一下，你用的gcc和clang是哪个版本吗？

2512500960, 2022-02-12:
> 还是不行。应该是编译器的问题。你能帮我看一下，你用的gcc和clang是哪个版本吗？ [image]
[image]

2512500960, 2022-02-12:
> > 没碰上过这个，网上搜这个错误，说是调错汇编器（as）了，gnu的as支持-EL选项。那就玄学操作一拨儿呗：
> > 
> > 1. ARCH=arm64 make clean，然后再make的时候ARCH=arm64家前面，额外再加一个LLVM_IAS=0，https://www.kernel.org/doc/html/latest/kbuild/llvm.html
> > 2. 全部删掉，重新来
> >    还是不行。找不到error，最后还是 `[Makefile:146: sub-make] Error 2`

之前说错了，写成V=1而不是V=s，来查看更详细的信息，尝试找出错误在哪儿。

Angel0726, 2022-02-12:
> > > 没碰上过这个，网上搜这个错误，说是调错汇编器（as）了，gnu的as支持-EL选项。那就玄学操作一拨儿呗：
> > > 
> > > 1. ARCH=arm64 make clean，然后再make的时候ARCH=arm64家前面，额外再加一个LLVM_IAS=0，https://www.kernel.org/doc/html/latest/kbuild/llvm.html
> > > 2. 全部删掉，重新来
> > >    还是不行。找不到error，最后还是 `[Makefile:146: sub-make] Error 2`
> 
> 之前说错了，写成V=1而不是V=s，来查看更详细的信息，尝试找出错误在哪儿。

好的

Angel0726, 2022-02-12:
> https://github.com/kdrag0n/proton-clang#as-unrecognized-option--el

谢谢

Angel0726, 2022-02-12:
> > > 没碰上过这个，网上搜这个错误，说是调错汇编器（as）了，gnu的as支持-EL选项。那就玄学操作一拨儿呗：
> > > 
> > > 1. ARCH=arm64 make clean，然后再make的时候ARCH=arm64家前面，额外再加一个LLVM_IAS=0，https://www.kernel.org/doc/html/latest/kbuild/llvm.html
> > > 2. 全部删掉，重新来
> > >    还是不行。找不到error，最后还是 `[Makefile:146: sub-make] Error 2`
> 
> 之前说错了，写成V=1而不是V=s，来查看更详细的信息，尝试找出错误在哪儿。
找到错误了。我总觉得clang有问题
[image]

2512500960, 2022-02-12:
> 没碰上过这个，网上搜这个错误，说是调错汇编器（as）了，gnu的as支持-EL选项。那就玄学操作一拨儿呗：
>
> 1. ARCH=arm64 make clean，然后再make的时候ARCH=arm64家前面，额外再加一个LLVM_IAS=0，https://www.kernel.org/doc/html/latest/kbuild/llvm.html
> 2. 全部删掉，重新来 [...]

可能是因为clang版本不一样？

Angel0726, 2022-02-12:
> > > > > > 没碰上过这个，网上搜这个错误，说是调错汇编器（as）了，gnu的as支持-EL选项。那就玄学操作一拨儿呗：
> > > > > > 
> > > > > > 1. ARCH=arm64 make clean，然后再make的时候ARCH=arm64家前面，额外再加一个LLVM_IAS=0，https://www.kernel.org/doc/html/latest/kbuild/llvm.html
> > > > > > 2. 全部删掉，重新来
> > > > > >    还是不行。找不到error，最后还是 `[Makefile:146: sub-make] Error 2`
> > > > 
> > > > 
> > > > 之前说错了，写成V=1而不是V=s，来查看更详细的信息，尝试找出错误在哪儿。
> > > > 找到错误了。我总觉得clang有问题
> > > > [image]
> > > > [image]
> > 
> > 
> > 可能是因为clang版本不一样？
> 
> clang-r383902b1、clang-r383902、clang-r383902c试过了不行。是不是gcc也有问题

你可以看一下你的版本吗？

2512500960, 2022-02-12:
我用的那个版本就是clang-r383902，docker我没试过。我试过增加usb无线网卡驱动，然后用手机抓wifi握手包，成功了的。

Angel0726, 2022-02-12:
> 你想在android上跑docker，是参照这一套来的吗？https://gist.github.com/FreddieOliveira/efe850df7ff3951cb62d74bd770dce27#221-general-compiling-instructions

关键是编译内核，这一步不成功

Angel0726, 2022-02-12:
> 我用的那个版本就是clang-r383902，docker我没试过。我试过增加usb无线网卡驱动，然后用手机抓wifi握手包，成功了的。

gcc用的什么

Angel0726, 2022-02-13:
> > > 我用的那个版本就是clang-r383902，docker我没试过。我试过增加usb无线网卡驱动，然后用手机抓wifi握手包，成功了的。
> > 
> > 你是用的wsl2还是原生linux系统？gcc用的哪个版本呀？是不是要加个32位的gcc编译器
> 
> 原生ubuntu20.10，gcc就是4.9，到了交叉编译阶段之后，用了clang之后不用gcc的吧，

我看hostgcc=gcc呀

2512500960, 2022-02-13:
> > > > 我用的那个版本就是clang-r383902，docker我没试过。我试过增加usb无线网卡驱动，然后用手机抓wifi握手包，成功了的。
> > > 
> > > 
> > > 你是用的wsl2还是原生linux系统？gcc用的哪个版本呀？是不是要加个32位的gcc编译器
> > 
> > 
> > 原生ubuntu20.10，gcc就是4.9，到了交叉编译阶段之后，用了clang之后不用gcc的吧，
> 
> 我看hostgcc=gcc呀

hostcc是用来编译scripts目录下面一些工具的。这个时候，既然没法儿直接定位分析问题，那就玄学一下，不用wsl，直截了当装一个干净的ubuntu20.04lts

Angel0726, 2022-02-13:
> > > > > 我用的那个版本就是clang-r383902，docker我没试过。我试过增加usb无线网卡驱动，然后用手机抓wifi握手包，成功了的。
> > > > 
> > > > 
> > > > 你是用的wsl2还是原生linux系统？gcc用的哪个版本呀？是不是要加个32位的gcc编译器
> > > 
> > > 
> > > 原生ubuntu20.10，gcc就是4.9，到了交叉编译阶段之后，用了clang之后不用gcc的吧，
> > 
> > 
> > 我看hostgcc=gcc呀
> 
> hostcc是用来编译scripts目录下面一些工具的。这个时候，既然没法儿直接定位分析问题，那就玄学一下，不用wsl，直截了当装一个干净的ubuntu20.04lts
这是编译的日志
[xm.log](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/files/8055073/xm.log)

2512500960, 2022-02-13:
> 我用的那个版本就是clang-r383902，docker我没试过。我试过增加usb无线网卡驱动，然后用手机抓wifi握手包，成功了的。
>
>
> 你是用的wsl2还是原生linux系统？gcc用的哪个版本呀？是不是要加个32位的gcc编译器
>
>
> 原生ubuntu20.10，gcc就是4.9，到了交叉编译阶段之后，用了clang之后不用gcc的吧，
>
>
> 我看hostgcc=gcc呀 [...]

这个编译日志里面有什么有意思的内容吗

Angel0726, 2022-02-13:
> 这个编译日志里面有什么有意思的内容吗

你是做安卓开发的吗？

Angel0726, 2022-02-15:
> > > > > > > > > 我用的那个版本就是clang-r383902，docker我没试过。我试过增加usb无线网卡驱动，然后用手机抓wifi握手包，成功了的。
> > > > > > > > 
> > > > > > > > 
> > > > > > > > 你是用的wsl2还是原生linux系统？gcc用的哪个版本呀？是不是要加个32位的gcc编译器
> > > > > > > 
> > > > > > > 
> > > > > > > 原生ubuntu20.10，gcc就是4.9，到了交叉编译阶段之后，用了clang之后不用gcc的吧，
> > > > > > 
> > > > > > 

修改了一个gcc，有了新进展。现在报错如下：

[image]

Angel0726, 2022-02-17:
> > > > > > > > > 我用的那个版本就是clang-r383902，docker我没试过。我试过增加usb无线网卡驱动，然后用手机抓wifi握手包，成功了的。
> > > > > > > > 
> > > > > > > > 
> > > > > > > > 你是用的wsl2还是原生linux系统？gcc用的哪个版本呀？是不是要加个32位的gcc编译器
> > > > > > > 
> > > > > > > 
> > > > > > > 原生ubuntu20.10，gcc就是4.9，到了交叉编译阶段之后，用了clang之后不用gcc的吧，
> > > > > > 
> > > > > > 
> > > > > > 我看hostgcc=gcc呀
> > > > > 
> > > > > 
> > > > > hostcc是用来编译scripts目录下面一些工具的。这个时候，既然没法儿直接定位分析问题，那就玄学一下，不用wsl，直截了当装一个干净的ubuntu20.04lts
> > > > > 这是编译的日志
> > > > > [xm.log](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/files/8055073/xm.log)
> > > 
> > > 
> > > 这个编译日志里面有什么有意思的内容吗
> > 
> > 
> > 你是做安卓开发的吗？
> 
> 不做
我看clang选择的gcc是x86_64是不是这个问题。选择的gcc不应该是arm的吗？
[image]

2512500960, 2022-02-17:
> 我用的那个版本就是clang-r383902，docker我没试过。我试过增加usb无线网卡驱动，然后用手机抓wifi握手包，成功了的。
>
>
> 你是用的wsl2还是原生linux系统？gcc用的哪个版本呀？是不是要加个32位的gcc编译器
>
>
> 原生ubuntu20.10，gcc就是4.9，到了交叉编译阶段之后，用了clang之后不用gcc的吧，
>
>
> 我看hostgcc=gcc呀 [...]

https://gitee.com/erlkonig/xiaomi_kernel_cezanne_r_oss/blob/cezanne-r-oss/kernel_compile_step.txt

链接：https://pan.baidu.com/s/1NiVglUt1oqFexoqdQ09EGQ 
提取码：mhhh

Angel0726, 2022-02-20:
> https://gitee.com/erlkonig/xiaomi_kernel_cezanne_r_oss/blob/cezanne-r-oss/kernel_compile_step.txt
>
> 链接：https://pan.baidu.com/s/1NiVglUt1oqFexoqdQ09EGQ
> 提取码：mhhh

谢谢，这个虚拟机里面安装了工具包吗？

2512500960, 2022-02-20:
> 谢谢，这个虚拟机里面安装了工具包吗？

虚拟机密码是ubuntu，里面已经安装好了依赖包，下载好了用到的编译工具链，也有K30的内核源码，都在家目录里面，步骤也写到里面一个txt文件了。

Angel0726, 2022-02-20:
十分感谢，我买个百度会员。我问一下你都会怎么找编译内核的相关资料的。我看仓库里对编译工具的版本跟编译步骤，都没有详细描述。

2512500960, 2022-02-20:
> 十分感谢，我买个百度会员。我问一下你都会怎么找编译内核的相关资料的。我看仓库里对编译工具的版本跟编译步骤，都没有详细描述。

阿里云盘没法儿分享压缩包，不然就用阿里云盘了，资料的话，直接google

Angel0726, 2022-02-21:
> > 十分感谢，我买个百度会员。我问一下你都会怎么找编译内核的相关资料的。我看仓库里对编译工具的版本跟编译步骤，都没有详细描述。
> 
> 阿里云盘没法儿分享压缩包，不然就用阿里云盘了，资料的话，直接google

嗯嗯，了解。关于资料，我看Documentation里面也没有。
[image]

2512500960, 2022-02-21:
> > > 十分感谢，我买个百度会员。我问一下你都会怎么找编译内核的相关资料的。我看仓库里对编译工具的版本跟编译步骤，都没有详细描述。
> > 
> > 
> > 阿里云盘没法儿分享压缩包，不然就用阿里云盘了，资料的话，直接google
> 
> 嗯嗯，了解。关于资料，我看Documentation里面也没有。 [image]

android内核编译起来和交叉编译普通的linux内核源码基本是一样。简单一点儿的情况的话，android11之后的android内核源码目录下面有一些build.config，是配合GKI使用的，参见 https://source.android.com/setup/build/building-kernels?hl=zh-cn 。 内核源码到底怎么编译，还是得看这份儿源码的厂家怎么整的了，厂家用的什么工具链什么编译环境，尽可能一样，不知道的话也只能靠猜了。

2512500960, 2022-02-21:
> android内核编译起来和交叉编译普通的linux内核源码基本是一样。简单一点儿的情况的话，android11之后的android内核源码目录下面有一些build.config，是配合GKI使用的，参见 https://source.android.com/setup/build/building-kernels?hl=zh-cn 。 [...]

小米的话，他的高通cpu那些内核源码应该用的是这个来编译？https://github.com/MiCode/kernel_build，这个从aosp里面的东西改过来的。

Angel0726, 2022-02-21:
> 

https://github.com/MiCode/MTK_kernel_modules/tree/cezanne-r-oss 这个是什么

2512500960, 2022-02-21:
> > 
> 
> https://github.com/MiCode/MTK_kernel_modules/tree/cezanne-r-oss 这个是什么

驱动，蓝牙，wifi，fm等的驱动。

Angel0726, 2022-02-21:
> > > 
> > 
> > 
> > https://github.com/MiCode/MTK_kernel_modules/tree/cezanne-r-oss 这个是什么
> 
> 驱动，蓝牙，wifi，fm等的驱动。

那如果只编译https://github.com/MiCode/Xiaomi_Kernel_OpenSource，不编译驱动手机会不会没法使用呀。

2512500960, 2022-02-21:
> 那如果只编译[https://github.com/MiCode/Xiaomi_Kernel_OpenSource，不编译驱动手机会不会没法使用呀。](https://github.com/MiCode/Xiaomi_Kernel_OpenSource%EF%BC%8C%E4%B8%8D%E7%BC%96%E8%AF%91%E9%A9%B1%E5%8A%A8%E6%89%8B%E6%9C%BA%E [...]

是的，wifi什么的没法儿用，因为内核模块和内核之间有个版本检验，一般都是要一起编译的。

Angel0726, 2022-02-21:
> 是的，wifi什么的没法儿用，因为内核模块和内核之间有个版本检验，一般都是要一起编译的。
我编译内核是想装docker。现在这意思是我还要编译装驱动了？

[34 more replies]