MiCode/Xiaomi_Kernel_OpenSource #2318: CAMELLIA-R-OSS IS MISSING THE TOUCHSCREEN DRIVER.

duelistsage, 2021-09-04:
PLEASE!!! add in camellia-r-oss source the corresponding touchscreen driver. @mi-code  
[image]
[image]
[camellia-r-oss](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/camellia-r-oss) @mi-code

duelistsage, 2021-09-04:
@mi-code

duelistsage, 2021-09-04:
[camellia-r-oss](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/camellia-r-oss) @mi-code

duelistsage, 2021-09-04:
[image]

willzyx-hub, 2021-09-04:
> [camellia-r-oss](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/camellia-r-oss) @mi-code

Can you fucking stop spamming ??? My Email is hurting me now.

duelistsage, 2021-09-04:
> > [camellia-r-oss](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/camellia-r-oss) @mi-code
> 
> Can you fucking stop spamming ??? My Email is hurting me now.

I won't fucking stop until I have the fucking driver!

ZIDAN44, 2021-09-05:
Xiaomi never see "open issue".. lmaoooo

willzyx-hub, 2021-09-05:
> Xiaomi never see "open issue".. lmaoooo

That's why he stupid, trying to spam Xiaomi it gonna be useless

Aquarius223, 2021-09-06:
In the screenshot you provided, the TP touch screen panel model is NT36672C, and the driver is in https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/camellia-r-oss/drivers/input/touchscreen/mediatek/NT36672C

willzyx-hub, 2021-09-06:
> In the screenshot you provided, the TP touch screen panel model is NT36672C, and the driver is in https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/camellia-r-oss/drivers/input/touchscreen/mediatek/NT36672C

This device use FT8722

KevinZonda, 2021-09-10:
It is **SO OFFENSIVE** that send plenty of meaningless issues. PLS close all of them.