MiCode/Xiaomi_Kernel_OpenSource #2319: There is no "DC Dimming" support in the current kernel sources of Xiaomi Mi 8 (dipper)

altirtix, 2021-09-11:
Hello dear developers.
Are there any plans for source updates for Xiaomi Mi 8 (dipper)?
The fact is that the current version of the kernel lacks support for some functions, for example "DC Dimming", which is very important.
Thank you for your reply, it is very important and valuable for users of this device.

Aquarius223, 2021-09-12:
There is dc dimming support in dipper-q-oss

closed 2022-10-14 by altirtix

altirtix, 2022-10-14:
> There is dc dimming support in dipper-q-oss

Hello. Can I ask you to remove this issue? The fact is that it is indexed by the search engine when someone tries to find me by nickname. This spoils my privacy. Thanks in advance.

Ki10V01T, 2025-09-17:
> There is dc dimming support in dipper-q-oss

Tell me please: how do I enable this feature in lineage os 22?