MiCode/Xiaomi_Kernel_OpenSource #2320: 编译K20 phoenix-r-oss warning: ISO C90 forbids mixing declarations and code

y0IkE, 2021-09-14:
CC      drivers/clk/clk-mux.o
../kernel/sched/fair.c:5975:17: warning: ISO C90 forbids mixing declarations and code [-Wdeclaration-after-statement]
error, forbidden warning: fair.c:5975
make[3]: *** [../scripts/Makefile.build:361：kernel/sched/fair.o] 错误 1
make[2]: *** [../scripts/Makefile.build:676：kernel/sched] 错误 2
make[1]: *** [/home/yyy/phoenix-r-oss/Makefile:1189：kernel] 错误 2
make[1]: *** 正在等待未完成的任务....

y0IkE, 2021-09-14:
上面的问题解决了，又冒出很多问题
../kernel/sched/fair.c:7776:21: error: use of undeclared identifier 'FULL_THROTTLE_BOOST'
                        sched_boost() != FULL_THROTTLE_BOOST ||
                                         ^
../kernel/sched/fair.c:8199:7: error: use of undeclared identifier 'sched_prefer_idle_on_input'
                if (sched_prefer_idle_on_input)
                    ^
../kernel/sched/fair.c:9374:3: error: implicit declaration of function 'trace_sched_load_balance_skip_tasks' [-Werror,-Wimplicit-function-declaration]
                trace_sched_load_balance_skip_tasks(env->src_cpu, env->dst_cpu,
                ^
../kernel/sched/fair.c:9374:3: error: this function declaration is not a prototype [-Werror,-Wstrict-prototypes]
../kernel/sched/fair.c:11170:5: error: implicit declaration of function 'mark_reserved' [-Werror,-Wimplicit-function-declaration]
                                mark_reserved(this_cpu);
                                ^
../kernel/sched/fair.c:11170:5: error: this function declaration is not a prototype [-Werror,-Wstrict-prototypes]
6 errors generated.

Morrigan-Ship, 2025-05-24:
same error

sumitd058, 2025-07-14:
for me also,

Anybody could please provide solution for this