MiCode/Xiaomi_Kernel_OpenSource #2335: crux-p-oss编译问题

2898353784, 2021-11-02:
按照Wiki流程走的
部署好环境后
make -j$(nproc) O=out REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_6.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu-
nan@nan-virtual-machine:~/crux-p-oss$ make -j$(nproc) O=out REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_6.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu-
make[1]: Entering directory '/home/nan/crux-p-oss/out'
  CHK     include/config/kernel.release
  GEN     ./Makefile
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  Using .. as source for kernel
  CHK     scripts/mod/devicetable-offsets.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    ../scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CHK     kernel/config_data.h
  DTC     arch/arm64/boot/dts/qcom/sm8150-rumi.dtb
  DTC     arch/arm64/boot/dts/qcom/sm8150-mtp.dtb
  DTC     arch/arm64/boot/dts/qcom/sm8150-cdp.dtb
  DTC     arch/arm64/boot/dts/qcom/sm8150-qrd.dtb
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1110.1-34 Label or path dsi_visionox_fhd_rm692a9_f1b_cmd not found
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1123.1-32 Label or path dsi_samsung_fhd_ea8076_f1s_cmd not found
FATAL ERROR: Syntax error parsing input tree
scripts/Makefile.lib:325: recipe for target 'arch/arm64/boot/dts/qcom/sm8150-rumi.dtb' failed
make[3]: *** [arch/arm64/boot/dts/qcom/sm8150-rumi.dtb] Error 1
make[3]: *** 正在等待未完成的任务....
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1110.1-34 Label or path dsi_visionox_fhd_rm692a9_f1b_cmd not found
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1123.1-32 Label or path dsi_samsung_fhd_ea8076_f1s_cmd not found
Error: ../arch/arm64/boot/dts/qcom/sm8150-mtp.dtsi:244.1-28 Label or path dsi_samsung_fhd_ea8076_cmd not found
FATAL ERROR: Syntax error parsing input tree
scripts/Makefile.lib:325: recipe for target 'arch/arm64/boot/dts/qcom/sm8150-mtp.dtb' failed
make[3]: *** [arch/arm64/boot/dts/qcom/sm8150-mtp.dtb] Error 1
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1110.1-34 Label or path dsi_visionox_fhd_rm692a9_f1b_cmd not found
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1123.1-32 Label or path dsi_samsung_fhd_ea8076_f1s_cmd not found
FATAL ERROR: Syntax error parsing input tree
scripts/Makefile.lib:325: recipe for target 'arch/arm64/boot/dts/qcom/sm8150-cdp.dtb' failed
make[3]: *** [arch/arm64/boot/dts/qcom/sm8150-cdp.dtb] Error 1
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1110.1-34 Label or path dsi_visionox_fhd_rm692a9_f1b_cmd not found
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1123.1-32 Label or path dsi_samsung_fhd_ea8076_f1s_cmd not found
FATAL ERROR: Syntax error parsing input tree
scripts/Makefile.lib:325: recipe for target 'arch/arm64/boot/dts/qcom/sm8150-qrd.dtb' failed
make[3]: *** [arch/arm64/boot/dts/qcom/sm8150-qrd.dtb] Error 1
../scripts/Makefile.build:678: recipe for target 'arch/arm64/boot/dts/qcom' failed
make[2]: *** [arch/arm64/boot/dts/qcom] Error 2
make[2]: *** 正在等待未完成的任务....
arch/arm64/Makefile:184: recipe for target 'dtbs' failed
make[1]: *** [dtbs] Error 2
make[1]: *** 正在等待未完成的任务....
make[1]: Leaving directory '/home/nan/crux-p-oss/out'
Makefile:146: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2

#### failed to build some targets (3 seconds) ####

dtc是Android9 源码里面的
nan@nan-virtual-machine:~/crux-p-oss$ dtc -v
Version: DTC 1.4.4-Android-build

2898353784, 2021-11-02:
不设置DTC_EXT=dtc的情况下结果如下

make[1]: Entering directory '/home/nan/crux-p-oss/out'
  CHK     include/config/kernel.release
  GEN     ./Makefile
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  Using .. as source for kernel
  CHK     scripts/mod/devicetable-offsets.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    ../scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CHK     kernel/config_data.h
  DTC     arch/arm64/boot/dts/qcom/apq8016-sbc.dtb
  DTC     arch/arm64/boot/dts/qcom/apq8096-db820c.dtb
  DTC     arch/arm64/boot/dts/qcom/ipq8074-hk01.dtb
  DTC     arch/arm64/boot/dts/qcom/msm8916-mtp.dtb
  DTC     arch/arm64/boot/dts/qcom/msm8992-bullhead-rev-101.dtb
  DTC     arch/arm64/boot/dts/qcom/msm8994-angler-rev-101.dtb
  DTC     arch/arm64/boot/dts/qcom/msm8996-mtp.dtb
  DTC     arch/arm64/boot/dts/qcom/sm8150-rumi.dtb
  DTC     arch/arm64/boot/dts/qcom/sm8150-mtp.dtb
  DTC     arch/arm64/boot/dts/qcom/sm8150-cdp.dtb
  DTC     arch/arm64/boot/dts/qcom/sm8150-qrd.dtb
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1110.1-34 Label or path dsi_visionox_fhd_rm692a9_f1b_cmd not found
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1123.1-32 Label or path dsi_samsung_fhd_ea8076_f1s_cmd not found
FATAL ERROR: Syntax error parsing input tree
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1110.1-34 Label or path dsi_visionox_fhd_rm692a9_f1b_cmd not found
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1123.1-32 Label or path dsi_samsung_fhd_ea8076_f1s_cmd not found
scripts/Makefile.lib:325: recipe for target 'arch/arm64/boot/dts/qcom/sm8150-rumi.dtb' failed
make[3]: *** [arch/arm64/boot/dts/qcom/sm8150-rumi.dtb] Error 1
make[3]: *** 正在等待未完成的任务....
Error: ../arch/arm64/boot/dts/qcom/sm8150-mtp.dtsi:244.1-28 Label or path dsi_samsung_fhd_ea8076_cmd not found
FATAL ERROR: Syntax error parsing input tree
scripts/Makefile.lib:325: recipe for target 'arch/arm64/boot/dts/qcom/sm8150-mtp.dtb' failed
make[3]: *** [arch/arm64/boot/dts/qcom/sm8150-mtp.dtb] Error 1
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1110.1-34 Label or path dsi_visionox_fhd_rm692a9_f1b_cmd not found
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1123.1-32 Label or path dsi_samsung_fhd_ea8076_f1s_cmd not found
FATAL ERROR: Syntax error parsing input tree
scripts/Makefile.lib:325: recipe for target 'arch/arm64/boot/dts/qcom/sm8150-cdp.dtb' failed
make[3]: *** [arch/arm64/boot/dts/qcom/sm8150-cdp.dtb] Error 1
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1110.1-34 Label or path dsi_visionox_fhd_rm692a9_f1b_cmd not found
Error: ../arch/arm64/boot/dts/qcom/sm8150-sde-display.dtsi:1123.1-32 Label or path dsi_samsung_fhd_ea8076_f1s_cmd not found
FATAL ERROR: Syntax error parsing input tree
scripts/Makefile.lib:325: recipe for target 'arch/arm64/boot/dts/qcom/sm8150-qrd.dtb' failed
make[3]: *** [arch/arm64/boot/dts/qcom/sm8150-qrd.dtb] Error 1
../scripts/Makefile.build:678: recipe for target 'arch/arm64/boot/dts/qcom' failed
make[2]: *** [arch/arm64/boot/dts/qcom] Error 2
arch/arm64/Makefile:184: recipe for target 'dtbs' failed
make[1]: *** [dtbs] Error 2
make[1]: *** 正在等待未完成的任务....
make[1]: Leaving directory '/home/nan/crux-p-oss/out'
Makefile:146: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2

#### failed to build some targets (4 seconds) ####

psndna88, 2021-11-03:
mu device haydn:for some devices the device tree files are incomplete, i faced similar in haydn-r-oss, i used the files from star-r-oss and further updated from odin-r-oss

2898353784, 2021-11-04:
It's strange that I compiled the same source code successfully once before, but later there was a problem compiling DTC on the new device. I tried different versions of DTC, and the final results were almost the same

2898353784, 2021-11-04:
> my device haydn: for some devices the device tree files are incomplete, i faced similar in haydn-r-oss, i used the files from star-r-oss and further updated from odin-r-oss
> 
> haydn-r-oss has lot of incomplete souces like missing fingerprint driver, broken touchscreen driver, missing charging driver (MI turbo charge doesnt work), voWiFi doesnt work

It's strange that I compiled the same source code successfully once before, but later there was a problem compiling DTC on the new device. I tried different versions of DTC, and the final results were almost the same

dajumaomao, 2023-03-23:
[image]
Same issue. Have you already found solutions?
OS: Ubuntu 20.04 WSL

dajumaomao, 2023-03-23:
> [image] Same issue. Have you already found solutions? OS: Ubuntu 20.04 WSL
 Got it. You all need to add this config in your defconfig file.
[image]