MiCode/Xiaomi_Kernel_OpenSource #2339: [LIME][REDMI 9T/POWER/M3] Release Android R LA.UM.9.15.r1 Source Code

gaganmalvi, 2021-11-12:
We require this. As we see you have released latest builds for China, please do the needful.

dragonpt, 2021-11-23:
Not only we need it due to GPL compliance, but also because there's issues with one of the Novatek's panel revision

javashin, 2022-01-15:
Release Sources

kalkioss, 2022-02-24:
It has been months since Android R release for lime. Kindly release the kernel sources as well.

salisa5147, 2022-03-09:
It has been months since Android 11 release for lime. Kindly release the kernel sources as well.

kalkioss, 2023-04-07:
Bump Xiaomi bump!!!!!!!!!!!!!!!!!!!!!!!!!!
Android 12 stable beta for lime is already released and will be released as stable very soon, atleast update the kernel sources now.
This is not just violation of GPL compliance but also about the broken touch on new touchpanels due to missing drivers.